wokwi/wokwi-features #320: Some utf8 throws off parser

muranyia, 2022-03-02:
**Describe the bug**
Some utf8 characters make the parser fail with error. It doesn't matter whether the incriminated characters are in code blocks or in comments.

**To Reproduce**
Put either character inside a string that the code uses or in a comment: ¢£¤¥¦
There might be more offending characters, I only tried these.

**Expected behavior**
With the exception of a few special utf8 characters, they should be usable in strings and tolerated in comments.

**Environment (please complete the following information):**
 - OS: [Linux]
 - Browser [Firefox]
 - Version [97.0]

muranyia, 2022-03-02:
[Here](https://wokwi.com/arduino/projects/325020837609472594) it is - the offending chars are in a comment now on line 10.

urish, 2022-03-02:
Uploaded a fix - can you please check now?

muranyia, 2022-03-02:
Works perfectly - thanks a lot!

closed 2022-03-02 by muranyia