wokwi/wokwi-features #321: Support for MAX6675 (high temp sensor) to be integrated with online simulations

alzin, 2022-03-07:
I have a suggestion to add the MAX6675 which is one of the famous and cheapest (high-temperature sensors) to be integrated and simulated with MUCUs like ESP32.

when I searched in the library manager I found that I can add a library that supports it, but still can`t show it in the simulation, so please consider adding it as a new feature!!

MAX6675 datasheet for reference:
[https://datasheets.maximintegrated.com/en/ds/MAX6675.pdf](url)

urish, 2022-03-07:
Thanks! Opened this feature request [for voting](https://wokwi.com/features?sort=newest#feature-1160983423)

hayschan, 2023-12-28:
Another workaround for this is to add MAX6675 as a custom chip using the [new API](https://docs.wokwi.com/chips-api/getting-started).

hayschan, 2025-05-20:
Update to this issue. 

> Another workaround for this is to add MAX6675 as a custom chip using the [new API](https://docs.wokwi.com/chips-api/getting-started).
> 
> The new custom chip API supports [SPI](https://docs.wokwi.com/chips-api/spi).

Here is the MAX6675 in Wokwi Simulator using custom chip API: https://wokwi.com/projects/389412304853763073