wokwi/wokwi-features #324: RPi Pico SDK interface, no hardware/pwm.h

eleciawhite, 2022-03-17:
When I use the [PWM example from the Pico SDK docs](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pwm.html#pwm_example), I get a compile time error about hardware/pwm.h not being found.

`
PICO_SDK_PATH is /home/pico/pico-sdk
PICO platform is rp2040.
PICO target board is pico.
Using board configuration from /home/pico/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /home/pico/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
/home/pico/project/src/main.c:7:10: fatal error: hardware/pwm.h: No such file or directory
    7 | #include "hardware/pwm.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/wokwi_project.dir/build.make:63: CMakeFiles/wokwi_project.dir/src/main.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1529: CMakeFiles/wokwi_project.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
`

[Here is my project](https://wokwi.com/projects/326401599629951572) to reproduce the error.

**Expected behavior**
I'm hoping the PWM features can be hooked to the Logic Analyzer component to look at how duty rate changes the trace and then how frequency changes the rate. And then look at those on an LED and then on a speaker. But at this point, just compiling would be great.

**Environment (please complete the following information):**
 - OS: Windows
 - Browser: Chrome
 - Version: 99.0.4844.51

urish, 2022-03-17:
Fixed by adding the hardware_pwm library to the Pico SDK [default CMakeLists.txt file](https://github.com/wokwi/wokwi-builders/blob/97aa8ccc8f7550549adfe3f9ed21668fa7bfd9d0/pico-sdk/project/CMakeLists.txt#L16).

Are there any other libraries that you need?

As far as the actual PWM feature - it's not implemented yet for the rp2040js library, but there's [an open issue](https://github.com/wokwi/rp2040js/issues/84) for it. Ideally, by what time would you need it for the class?

eleciawhite, 2022-03-17:
Mar 27 would be the last day it I could put it in a lecture.  If it isn't likely to be done by then, let me know and I'll figure something else out.

For other libraries... actually, let me go by Pico Examples... in order of desirability:
 * [UART](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__uart.html#uart_example) <== **Already works**
 * [PWM](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pwm.html#pwm_example)
 * [Timer](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__pwm.html#pwm_example)
 * [ADC](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__adc.html#adc_example)
 * [I2C](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__i2c.html#i2c_example)
 * [Watchdog](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__watchdog.html#watchdog_example)
 
 (I'd ask for the [Flash Programming example](https://raspberrypi.github.io/pico-sdk-doxygen/group__hardware__flash.html#flash_example) but I suspect that is remarkably unsafe for a simulation.)

urish, 2022-03-17:
PWM - Noted. I'll look into it and get back to you at the beginning of next week with an answer.

Other libraries - added ADC, I2C and Flash. I think that the Flash Programming example should work (MicroPython uses it to manage its flash filesystem, and it works there).

For ADC - you can use the [potentiometer](https://docs.wokwi.com/parts/wokwi-potentiometer) or [slide potentiometer](https://docs.wokwi.com/parts/wokwi-slide-potentiometer) as an input device, or for added fun there's also an [Analog Joystick](https://docs.wokwi.com/parts/wokwi-analog-joystick)

Watchdog - the example compiles, but the peripheral itself [isn't there yet](https://docs.wokwi.com/parts/wokwi-pi-pico#simulation-features).

eleciawhite, 2022-03-18:
@urish: I thought about organizing the lecture and I can do the PWM stuff in physical HW and then assign the Wokwi project as part of a homework... that gives you an extra week if it makes this feature more viable.

urish, 2022-03-20:
Thanks @eleciawhite! 

I looked into this, and it seems like we can have PWM for Mar 27, and possibly even before. I will keep you posted!

closed 2022-03-22 by urish