wokwi/wokwi-features #325: Spanish Translation

tecnoblasco, 2022-03-17:
Hi,
here is my Spanish(European) translation.
Hope it helps.

[es.zip](https://github.com/wokwi/wokwi-features/files/8289183/es.zip)

urish, 2022-03-26:
Thank you! 

We already have a Spanish (eu) translation, but it had some missing texts, so I updated them based on your file. The update is live at [http](https://wokwi.com/projects/new?lang=es-ES):

There are some differences between the translation that you sent and the original translation by @angelmicelti (from #232), so I'd love to have your feedback regarding which version is better. The lines in red are the original translation, and the lines in green are the new (proposed) version:

```diff
-  "userMenu.signIn": "Entrar",
+  "userMenu.signIn": "Acceder",
-  "userMenu.theClub": "El Club",
+  "userMenu.theClub": "The Club",
-  "userMenu.featureRoadmap": "Hoja de ruta",
+  "userMenu.featureRoadmap": "Mejoras Futuras",
-  "project.header.likeProjectTooltip": "Me gusta este proyecto",
+  "project.header.likeProjectTooltip": "Dale 'Like' a este proyecto",
-  "project.header.unlikeProjectTooltip": "No me gusta este proyecto",
+  "project.header.unlikeProjectTooltip": "Dale 'Unlike' este proyecto",
-  "project.menu.saveACopy": "Guarda una copia",
+  "project.menu.saveACopy": "Guardar una copia",
-  "project.menu.lockProject": "Bloquear proyecto",
+  "project.menu.lockProject": "Bolquear proyecto",
-  "project.menu.unlistProject": "Quitar proyecto de la lista",
+  "project.menu.unlistProject": "Deslistar proyecto",
-  "project.menu.downloadProjectZip": "Descargar como archivo ZIP",
+  "project.menu.downloadProjectZip": "Descargar el proyecto en archivo ZIP",
-  "project.error.saveFailed": "No se puede guardar: proyecto bloqueado",
+  "project.error.saveFailed": "Fallo en el guardado: ¡El proyecto está bloqueado!",
-  "project.error.signInToLikeProject": "Por favor, ingresa para darle Like a este proyecto",
+  "project.error.signInToLikeProject": "Por favor acceda para poder hacer 'Like' a este proyecto.",
-  "editor.fileMenu.newFile": "Nuevo archivo...",
+  "editor.fileMenu.newFile": "Archivo Nuevo...",
-  "editor.fileMenu.confirmDelete": "Esto eliminará {{filename}}. ¿Seguir?",
+  "editor.fileMenu.confirmDelete": "Esta acción borrará {{filename}}. ¿Desea contiuar?",
-  "editor.fileMenu.error.invalidFileName": "Nombre de archivo incorrecto: no se permite el carácter {{char}} .",
+  "editor.fileMenu.error.invalidFileName": "Nombre de archivo no válido: el caracter {{char}} no está permitido.",
-  "editor.fileMenu.error.invalidFileType": "Tipo de archivo no válido. Archivos soportados: {{types}}.",
+  "editor.fileMenu.error.invalidFileType": "Nombre de archivo no válido. Archivos permitidos: {{types}}.",
-  "editor.fileMenu.error.uploadFailed": "Algunos archivos no pudieron subirse",
+  "editor.fileMenu.error.uploadFailed": "¡Alguno de los archivos no pudieron subirse!",
-  "diagram.menu.zoomIn": "Zoom hacia dentro",
+  "diagram.menu.zoomIn": "Acercar Zoom",
-  "diagram.menu.zoomOut": "Zoom hacia fuera",
+  "diagram.menu.zoomOut": "Alejar Zoom",
-  "diagram.menu.toggleGrid": "Alternar rejilla",
+  "diagram.menu.toggleGrid": "Activar/Desactivar la rejilla",
-  "diagram.menu.exitFullScreen": "Salir de pantalla completa",
+  "diagram.menu.exitFullScreen": "Salir de pantallla completa",
-  "diagram.addPart.wokwi-slide-switch": "Interruptor",
+  "diagram.addPart.wokwi-slide-switch": "Interruptor deslizante",
-  "diagram.addPart.wokwi-membrane-keypad": "Teclado",
+  "diagram.addPart.wokwi-membrane-keypad": "Teclado numérico",
-  "diagram.addPart.wokwi-logic-analyzer": "Analizador lógico (8 canales)",
+  "diagram.addPart.wokwi-logic-analyzer": "Analizador Lógico(8 canales)",
-  "diagram.addPart.wokwi-neopixel-meter": "Medidor NeoPixel",
+  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
-  "diagram.addPart.wokwi-slide-potentiometer": "Potenciómetro lineal",
+  "diagram.addPart.wokwi-slide-potentiometer": "Potenciómetro Deslizante",
-  "diagram.addPart.wokwi-servo": "Servo",
+  "diagram.addPart.wokwi-servo": "Servomotor",
-  "diagram.addPart.wokwi-rotary-dialer": "Marcador rotatorio",
+  "diagram.addPart.wokwi-rotary-dialer": "Dial Rotativo",
-  "diagram.addPart.wokwi-ssd1306": "SSD1306 OLED display",
+  "diagram.addPart.wokwi-ssd1306": "Pantalla SSD1306 OLED",
-  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Sensor de distancia ultrasónico",
+  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Sensor de Distancia por Ultrasonidos",
-  "diagram.addPart.wokwi-rgb-led": "LED RGB",
+  "diagram.addPart.wokwi-rgb-led": "RGB LED",
-  "diagram.addPart.wokwi-7segment": "Display de siete segmentos",
+  "diagram.addPart.wokwi-7segment": "Display de Siete Segmentos",
-  "diagram.addPart.wokwi-7segment-4digits": "Display de 7 segmentos (4 dígitos)",
+  "diagram.addPart.wokwi-7segment-4digits": "SDisplay de Siete Segmentos (4 dígitos)",
-  "diagram.addPart.wokwi-led-bar-graph": "Barra de LEDes",
+  "diagram.addPart.wokwi-led-bar-graph": "LED Bar Graph",
-  "diagram.addPart.wokwi-analog-joystick": "Joystick Analógico",
+  "diagram.addPart.wokwi-analog-joystick": "Joystick Analógico ",
-  "diagram.addPart.wokwi-ir-receiver": "Receptor IR",
+  "diagram.addPart.wokwi-ir-receiver": "IR Receptor",
-  "diagram.addPart.wokwi-ir-remote": "Mando IR",
+  "diagram.addPart.wokwi-ir-remote": "IR Mando a distancia",
-  "diagram.addPart.wokwi-pir-motion-sensor": "Sensor de movimiento PIR",
+  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Sensor de Movimiento",
-  "diagram.addPart.wokwi-ntc-temperature-sensor": "Sensor de temperatura analógico (NTC)",
+  "diagram.addPart.wokwi-ntc-temperature-sensor": "Resistenca variable con la Temperatura(NTC)",
-  "diagram.addPart.wokwi-74hc595": "74HC595 Registro de desplazamiento",
+  "diagram.addPart.wokwi-74hc595": "74HC595 Shift Register",
-  "diagram.addPart.wokwi-photoresistor-sensor": "Sensor de luz (LDR)",
+  "diagram.addPart.wokwi-photoresistor-sensor": "Fotoresistenciaa (LDR) Sensor",
-  "diagram.addPart.wokwi-mpu6050": "MPU6050 Acelerómetro + Giróscopo",
+  "diagram.addPart.wokwi-mpu6050": "MPU6050 Accelerómetro + Giroscopio",
```

tecnoblasco, 2022-03-31:
Hello,
I'm glad to be helpful.

I compared the original translation, and both look good to me. The differences are minimal.
While reviewing I detected some typos. I am attaching to this post  the corrected file.

Thank you for the development and maintenance of this project. I really appreciate it.

Regards,

[es-ES.zip](https://github.com/wokwi/wokwi-features/files/8391804/es-ES.zip)