wokwi/wokwi-features #326: DotStars / SK9822 / APA102 simulation

dodohand, 2022-03-19:
Very similar to the existing NeoPixel device, I'd like there to be a DotStar / SK9822 / APA102 device simulation.

I imagine that much of the NeoPixel graphics/display implementation can be carried over, and that the new, clocked bitstream ought not be too bad to implement in simulation.

Datasheets are available from adafruit: 
- https://cdn-learn.adafruit.com/assets/assets/000/084/591/original/APA102_LED.pdf?1574117503
- https://cdn-learn.adafruit.com/assets/assets/000/084/592/original/APA102_White_LED.pdf?1574117639
- https://cdn-learn.adafruit.com/assets/assets/000/084/593/original/APA102-2020_SMD_LED.pdf?1574117728

sutaburosu, 2022-03-19:
There is already a beta of APA102 type addressable LEDs. See [this project for an example of how to conjure them into your own project](https://wokwi.com/projects/309664470325527104).

urish, 2022-04-09:
Should be addressed as part of #240

closed 2022-04-09 by urish