wokwi/wokwi-features #332: Delete File option is missing on some files in editor

drf5n, 2022-04-01:
One should be able to delete mistaken uploads.

I uploaded a file to https://wokwi.com/projects/327748577668366932 and then tried to delete it.  The option was not present on the drop-down menu:

[image]

I expected the delete file command that is present for other tabs to be present on the uploaded file. For instance, here's one on diagram.json:

[image]
 
**Environment (please complete the following information):**
 - OS: Mac OSX
 - Browser: Chrome
 - Version 99.0.4844.83 (Official Build) (x86_64)

urish, 2022-04-04:
Thanks for reporting! Can you please check again?

drf5n, 2022-04-07:
I cannot replicate.  It seems to be OK now.  

I uploaded an additional file to that project, and was able to delete it both before saving the project, after saving the project, and after reloading the project with the saved file.

closed 2022-04-07 by urish