wokwi/wokwi-features #333: Add GLCD (KS0108) support

minolmal, 2022-04-06:
**Describe the solution you'd like**
I really loved the ability to simulate SSD1306 OLED displays before I upload to MCU. Prevents a lot of unnecessary write to MCU due to trial and error procedures. 
I would like to see the same approach used with KS0108 based mono colored GLCDs. 

[image]
[image]

the very basic functionality should be described in here.
https://www.electronicwings.com/sensors-modules/glcd-128x64

urish, 2022-04-06:
Thanks for the suggestion! I opened this request [for voting](https://wokwi.com/features?sort=newest#feature-1194296434)

urish, 2022-10-10:
Closing as this didn't get any votes in the past 6 months. 

Note that you could still try to create this part yourself using the new custom chip API prototype - see #301 for details.

closed 2022-10-10 by urish