wokwi/wokwi-features #334: Add Rust support

dcabanis, 2022-04-07:
**Describe the solution you'd like**
I think Rust is a very elegant solution for embedded programming. The Pi-Pico/Arduino boards have existing support in Rust. I would expect it to be fairly simple to launch cargo build instead of GCC to create a HEX file before loading it onto the simulator.

urish, 2022-04-07:
Thanks for the request!

There's some preliminary support for Rust on the ESP32: 
- [ESP32-C3 Rust Blinky](https://wokwi.com/projects/327576070332088915)
- [ESP32 Rust PWM](https://wokwi.com/projects/323765044769718868)

Alternatively, you can create the .HEX file on your computer (using whatever toolchain you want), then go to the Wokwi code editor, press "F1" and choose "Load HEX File and Start Simulation...".

Does this help?

dcabanis, 2022-04-07:
Thank you Uri, I will have a look at the links you provided. I just
finished watching your online class on the pi-pico. Although I am very
familiar with the Cortex-M0+ SW development you made this very engaging. I
particularly liked the GDB part. In this topic I have been  using a GDB
init file that I created years ago to dump on the console useful Core
information each time you execute a new command. It's tailored for the
cortex-m. If you are interested just tell me, I can send you the file over.
You only need to copy it in the directory you want to invoke GDB for it to
work.

urish, 2022-04-09:
Thanks for the feedback! Yeah, please share you GDB init file.

dcabanis, 2022-04-21:
Hello Uri,

Sorry for the delay, I went on holiday and I forgot about everything.
Here is the .gdbinit file that I use. I mainly use it for Assembly
development, so it might need tuning if you want to use it to debug
higher-level languages.

Enjoy.

David.

PS. I hope you are Okay in Israel at the moment I see in the news that
things are heating up.

-- 
Les sages sont instruits par la raison, les esprits moyens par l
'expérience, le stupide par la nécessité et la brute par l'instinct.

Cicero.

################################################################################
#
# Common gdb rules for Armv7-M
#
################################################################################

# Project specific commands
#set complaints 1
set language auto
set output-radix 0x10
set input-radix 0x10
set endian little

dir .

################################################################################
# Common commands to put in the home directory
define bpl
info breakpoints
end
document bpl
List breakpoints
end

define bp
set $SHOW_CONTEXT = 1
break * $arg0
end
document bp
Set a breakpoint on address
Usage: bp addr
end

define bpc
clear $arg0
end
document bpc
Clear breakpoint at function/address
Usage: bpc addr
end

define bpe
enable $arg0
end
document bpe
Enable breakpoint #
Usage: bpe num
end

define bpd
[... 318 lines cut ...]
    printf "Uses MSP 0x%x return.\n", $MSP
    set $armex_base = $MSP
    else
    printf "Uses PSP 0x%x return.\n", $PSP
    set $armex_base = $PSP
    end
  
    printf "xPSR            0x%x\n", *($armex_base+28)
    printf "ReturnAddress   0x%x\n", *($armex_base+24)
    printf "LR (R14)        0x%x\n", *($armex_base+20)
    printf "R12             0x%x\n", *($armex_base+16)
    printf "R3              0x%x\n", *($armex_base+12)
    printf "R2              0x%x\n", *($armex_base+8)
    printf "R1              0x%x\n", *($armex_base+4)
    printf "R0              0x%x\n", *($armex_base)
    printf "Return instruction:\n"
    x/i *($armex_base+24)
    printf "LR instruction:\n"
    x/i *($armex_base+20)
end
  
document armex
ARMv7 Exception entry behavior.
xPSR, ReturnAddress, LR (R14), R12, R3, R2, R1, and R0
end

dcabanis, 2022-04-21:
Uri,

Thanks for the links, its very exciting. I think the first link you
provided (Blinky) is taking me to an arduino sketch instead of a main.rs
file.
Do you have an alternative link I could look at?

Kindly,

David.

urish, 2022-04-21:
Sorry, my bad. Here's the rust one: https://wokwi.com/projects/325483260301804114

closed 2022-05-19 by urish

dcabanis, 2022-10-11:
Many Thanks Uri.

David.

urish, 2022-10-11:
You're welcome. We now also have a page dedicated for rust examples: https://wokwi.com/rust

dcabanis, 2022-10-11:
Absolutely brilliant!

Thank you very much.

David.