wokwi/wokwi-features #336: Potentiometer slider doesn;t appear until Simualtion is started

matititam, 2022-04-10:
To slider, won't appear in the potentiometer until I run the simulation

Link to the project here to reproduce: 
https://wokwi.com/projects/327381547863769683
[image]

Tio: seems to be that the attribute value should not be more than 1000. When a value less than 1000 is entered, the simulation reappears.

urish, 2022-04-12:
Thanks for reporting! Can you please check now?

eternaleclipse, 2022-04-14:
Works for me now in Firefox / Chrome (Firefox still has the slider moving bug (https://github.com/wokwi/wokwi-elements/issues/87) during simulation but it's not related).

closed 2022-04-14 by urish