wokwi/wokwi-features #338: External power source

eternaleclipse, 2022-04-13:
It would be nice to have an external power source component with configurable voltage.

Can be useful for playing with different voltages, and also for simulating circuits with different power sources (ex. Arduino controlling 220v power via relay).

urish, 2022-04-21:
The thing is, Wokwi does not simulate voltages. That being said, I'm seeing users creating a simulation with just electronics (and now also with custom chips too, see #301) and using the ATtiny85 as the "power source" for their circuit. So it might make sense to have some sort of a Power Source component that will be used for these situation.

I'm just not sure what this Power source look be like. Any suggestions?

eternaleclipse, 2022-04-23:
Yeah, I assume it will mainly be used to simulate circuits with relays (or something that makes sense IRL to use a power source other than what the main board provides).

It can look like a battery, with 2 nodes for + and -, and a user setting for voltage.

HeathersZen, 2024-10-20:
I just joined the club today, and instantly found a need for this feature. Designing actual circuits beyond the 3.3 the ESP32 provides means there must be a power supply somewhere...