wokwi/wokwi-features #339: Simple relay module with indicator LEDs

eternaleclipse, 2022-04-13:
It would be nice to have a simple relay module with indicator LEDs:
https://arduinogetstarted.com/tutorials/arduino-relay

There is currently an 8-pin [DPDT switch](https://docs.wokwi.com/parts/wokwi-ks2e-m-dc5), but for most circuits it's unintuitive / overkill, and doesn't provide visual indication of when it's on / off.

urish, 2022-04-14:
Thanks! Opened for [voting](https://wokwi.com/features?sort=newest#feature-1203598486)

arduino12, 2022-04-18:
Just voted for this!
1. Will be great to add ```channel_count``` attribute that will accept 1, 2, 4, 8, 16 like the **[common relay modules](https://www.aliexpress.com/item/1005001700441473.html)**.
2. Will be great to add ```active_polarity``` attribute that will accept a bool like set in those **[relay modules](https://www.aliexpress.com/item/1005002617199944.html)** jumpers.

Thanks @urish !

scorninpc, 2022-07-08:
I've been used this as a simple led. Just change the layout to a relay will be nice! 

+1

yanickrochon, 2022-07-30:
I have a project with relays, and having this component would greatly help!

urish, 2022-10-23:
This is getting up the vote list, so we may start to work on it soon

AriellaE, 2022-11-03:
Thanks to everyone who voted for this feature!

Here you go:
- [Sample project](https://wokwi.com/projects/347309345158988371)
- [Docs](https://docs.wokwi.com/parts/wokwi-relay-module)

Enjoy!

closed 2022-11-03 by AriellaE