wokwi/wokwi-features #340: Component properties pane

eternaleclipse, 2022-04-13:
**Is your feature request related to a problem? Please describe.**
It is currently only possible to change component properties by editing `diargams.json`.

**Describe the solution you'd like**
Add a pane for editing component properties, possibly using pickers with known values for each field for easier usability.

urish, 2022-04-21:
Thanks! Actually, been thinking to implement it as part of working on #244, as it seems related. How about a showing a cogwheel next to the selected element, and clicking that cogwheel will open a popup menu with the pickers?

eternaleclipse, 2022-04-24:
It's a matter of personal preference but IMO a better option would be a traditional "properties" pane present in IDEs like Visual Studio. I like the feeling of clicking on some component and instantly getting metadata about it. An additional benefit to this approach is that it would enable changing a property to the same value for multiple selected components.