wokwi/wokwi-features #342: UI: Multi-select, copy-paste

eternaleclipse, 2022-04-13:
When creating large circuits, it would be useful to have an option to multi-select (maybe rectangle with shift+drag mouse) and copy-paste / delete multiple components.

It can save time especially when creating multiple components with non-default properties (ex. resistor arrays).

urish, 2022-04-21:
Other than resistor/LED/button arrays, do you have other examples?

For arrays, it might be more useful to have an "array clone" function, where you can make a copies multiple copies of a single part with equal spacing

eternaleclipse, 2022-04-24:
Anything involving large repetitive segments - I can't think of other examples currently but as the platform continues to evolve, I assume there will be support for multiple MCUs, breadboards and these scenarios will become more common.

urish, 2022-08-23:
We now have both [multi select](https://docs.wokwi.com/guides/diagram-editor#selecting-multiple-parts) and [copy paste](https://docs.wokwi.com/guides/diagram-editor#copying-and-pasting-parts)!

closed 2022-08-23 by urish