wokwi/wokwi-features #345: Circuit Photo Download

bmeisels, 2022-04-15:
I would like a feature where one could download a picture of the circuit (kind of like a screenshot but organized and more pretty).
It would also label parts and wires based on the diagram.json.

urish, 2022-04-15:
Thanks for the suggestion! Do you have any sample / reference to what it might look like?

bmeisels, 2022-04-17:
The closest thing I can think of is the export to photo function of Fritzing

urish, 2022-04-17:
As far as I can tell, Fritzing exports an image of the diagram which includes what you see on the screen and their logo, without any additional labels. 

What's the use case for this feature?
Do we need to support exporting an image of running simulation (i.e. with a specific state), or just the schema in pre-simulation state?

bmeisels, 2022-04-18:
>As far as I can tell, Fritzing exports an image of the diagram which includes what you see on the screen and their logo, without any additional labels.

Exactly. I would like a better version of this in Wokwi.

>What's the use case for this feature?

I would like to have nice circuit diagrams when teaching embedded computing topics.

>Do we need to support exporting an image of running simulation (i.e. with a specific state), or just the schema in pre-simulation state?

No just the pre-simulation state.

urish, 2022-04-18:
Thanks! Opened for [voting](https://wokwi.com/features?sort=newest#feature-1205500502)

gmeader, 2022-04-27:
Does a screen capture (provided by all OS) not provide what is required?

LEKPHET-KUNSUE, 2024-01-24:
@urish! Look into your E-Mail!