wokwi/wokwi-features #347: Basic weatherstation sensors

martinius96, 2022-04-17:
I would like to suggest new feature for basic and very popular weatherstation (meteo) sensors, that should not be missing on Wokwi.

- Dallas DS18B20 with OneWire bus
- BME or BMP sensor on I2C bus
   - 180 or 280 version
- SHT2X or SHT3X

urish, 2022-04-17:
Hi Martin, thanks for the suggestion!

There's already an open request for the DS18B20, see #110.

BME/BMP/SHT2X/3X - do you have any information on which specific model is most popular? I can open it for voting, but from my experience, people usually vote for what's popular / what they use.

martinius96, 2022-04-18:
Oh, i missed that, DS18B20 was requested, sry.

I think BME280 will be ideal, because it is same as BMP280, that means it can measure temperature and atmospheric pressure, but it can also measure humidity (that BMP can't). Most users use BME280 nowadays (BME180 or BME085 are old...).

There are 3V3 and 5V versions of BME280 sensors, so I am not sure if that should be implemented too (maybe as some diagram.json parameter, also with changeable I2C address that can be changed via pullup / pulldown SD0 pin?). 

From SHT sensors, there are very minor differences between SHT 2X, 3X and 4X series.
Real difference I see only at operation voltage. SHT31 have input voltage from 2.4 V up to 5.5 V. SHT21 from 2.1 V up to 3.6 V and  SHT 40 from 1 V up to 3.6 V. Temperature range is identical, also accurancy. At humidity range is identical, there are minor difference in accurancy like from 1.5 %  up to 2%.

SHT31 is very popular, also it is relatively cheap and based on voltages it can be used with 5V Arduino and also 3V3 ESP32.

urish, 2022-04-18:
Thanks for the feedback! I updated the title of the issue accordingly and opened it for [voting](https://wokwi.com/features?sort=newest#feature-1206477581)

chconil, 2024-07-18:
Not tested, but found this 
https://github.com/bonnyr/wokwi-bme280-custom-chip/tree/main/chip

neela-rm, 2024-10-06:
Is there any rain sensors in wokwi?