wokwi/wokwi-features #349: Diode component (for making a matrix of switches)

gmeader, 2022-04-27:
I'd like a diode component - needed to create a matrix (rows and columns) of pushbuttons, where more than one button can be active at the same time. 

(I'm using the keypad matrix for now, but I need a different configuration - different numbers of switches placed in different locations. 
I guess an alternative to implementing a diode would be to add a keypad that is visually separable into individual switches with a configurable number of rows and columns driving the matrix.

**Here's info about this use case:**
[(https://arduino.stackexchange.com/questions/66691/which-kind-of-diodes-i-should-use-in-buttons-keys-matrix-input-making-gaming-k](https://arduino.stackexchange.com/questions/66691/which-kind-of-diodes-i-should-use-in-buttons-keys-matrix-input-making-gaming-k)

urish, 2022-04-27:
Thanks! I opened this for [voting](https://wokwi.com/features?sort=newest#feature-1216730994).

As far as I can tell, the Keypad Matrix does not have any diodes, so you'd still need the diodes even if you used it.

GModal, 2023-12-19:
I've written a Digitech FS300 footswitch - to - pi pico MIDI controller, and a diode would be very helpful. The code works, but could use a little refinement.

The FS300 uses a stereo cable (tip-ring-gnd), two diodes and three foot switches...the logic separating three switches ->into-> two GPIO inputs cannot be simulated without the diodes.

Phetdanai, 2023-12-22:
It's also for relay controlling.
Mr.Dong Garmon.

Phetdanai, 2023-12-29:
With a Motor

Phetdanai, 2023-12-29:
Digital GPIO pins on ESP32, Raspberry Pi Pico has voltage 3.3 volts
Unlike Arduino digital pins it has no more than 5 volts

Phetdanai, 2024-01-02:
```shell
curl --help
```

urish, 2024-01-02:
Hi @Phetdanai, you keep adding comments that are not related to the topic of the issue. Can you please stop?

campidelli, 2024-08-26:
That would be great!

I tried it with LEDs, which are essentially diodes, but it didn't work:

Not working (with LEDs) https://wokwi.com/projects/407280521016985601
Not working (with LEDs inverted) https://wokwi.com/projects/407280651790158849
Working (without LEDs) https://wokwi.com/projects/407281782665874433

I would like to scale this up to use a shit register together, like the following, but it doesn't work :(
https://wokwi.com/projects/407277725732051969

drf5n, 2024-10-08:
> That would be great!
>
> I tried it with LEDs, which are essentially diodes, but it didn't work:
>
> Not working (with LEDs) https://wokwi.com/projects/407280521016985601 Not working (with LEDs inverted) [...]

Thanks for the demos.

Here's another, much larger non-working diode attempt which also doesn't work:

https://wokwi.com/projects/386828635161861121

Here's a working one built with custom chips:

https://wokwi.com/projects/410325174113969153  per https://discord.com/channels/787627282663211009/1289692132776476853/1289786828915740704

nforystek, 2026-02-02:
I tried semi-extensive to do this with custom chips, at first 12 of them it failed faithfully.
https://wokwi.com/projects/454438811343692801 does work using just one custom chip.
Here is the project I was trying to mimic https://everycircuit.com/circuit/6148761842352128
So, I concur those results., it sounds familiar.  I would vote for this issue if I had the power!