wokwi/wokwi-features #353: adding more than one file with wrong order will lead into errors

willie68, 2022-05-02:
If you upload an already implemented project into wokwi containing more than one file, (in my case there are 2 step_2.ino and a prgmode.ino) the order is relevant: (In the normal Arduino.ide the order of the different files is irrelevant) 
If there are dependencies between the files, the order on wokwi is relevant, because if the order is not correct, wokwi will thro errors like: 

```
prgmode.ino: In function 'void prgDemoPrg()':
prgmode.ino:31:16: error: 'EEPROM' was not declared in this scope
   byte value = EEPROM.read(0);
                ^~~~~~
prgmode.ino:31:16: note: suggested alternative: 'EEPROM_h'
   byte value = EEPROM.read(0);
                ^~~~~~
                EEPROM_h
prgmode.ino: In function 'void programMode()':
prgmode.ino:41:22: error: 'SW_PRG' was not declared in this scope
   while (digitalRead(SW_PRG) == 0) {
                      ^~~~~~
prgmode.ino:41:22: note: suggested alternative: 'SPR0'
   while (digitalRead(SW_PRG) == 0) {
                      ^~~~~~
                      SPR0
prgmode.ino:46:3: error: 'addr' was not declared in this scope
   addr = 0;
   ^~~~
prgmode.ino:46:3: note: suggested alternative: 'doAddr'
   addr = 0;
   ^~~~
   doAddr
prgmode.ino:59:19: error: 'EEPROM' was not declared in this scope
     byte Eebyte = EEPROM.read(addr);
                   ^~~~~~
prgmode.ino:59:19: note: suggested alternative: 'EEPROM_h'
     byte Eebyte = EEPROM.read(addr);
                   ^~~~~~
                   EEPROM_h
prgmode.ino:68:23: error: 'SW_SEL' was not declared in this scope
       if (digitalRead(SW_SEL) == 0) {
                       ^~~~~~
prgmode.ino:68:23: note: suggested alternative: 'IVSEL'
       if (digitalRead(SW_SEL) == 0) {
                       ^~~~~~
                       IVSEL
prgmode.ino:75:24: error: 'SW_PRG' was not declared in this scope
     while (digitalRead(SW_PRG) == 1);
                        ^~~~~~
prgmode.ino:75:24: note: suggested alternative: 'SPR0'
     while (digitalRead(SW_PRG) == 1);
                        ^~~~~~
                        SPR0
prgmode.ino:83:23: error: 'SW_SEL' was not declared in this scope
       if (digitalRead(SW_SEL) == 0) {
                       ^~~~~~
prgmode.ino:83:23: note: suggested alternative: 'IVSEL'
       if (digitalRead(SW_SEL) == 0) {
                       ^~~~~~
                       IVSEL
Error during build: exit status 1

```

urish, 2022-05-02:
The reason this happens is how Arduino works: it looks for a .ino file whose name matches the name of the current directory, and uses that as the main file.

Since Wokwi doesn't have directories, it uses a slightly difference approach: it assumes the first .ino file in the project is the main file. Right now, the solution would be to remove `prgmode.ino` and upload it again, so it goes after `step_2.ino`. Once we have the ability to re-order the tabs (#207), this will even be easier to fix.

closed 2022-05-02 by urish

willie68, 2022-05-03:
Or other possibility:
you scan thru all ino files looking for one containing setup() and loop()...

urish, 2022-05-03:
That's a possibility, though, it's not always guaranteed that the main sketch will include setup() and loop(). You can even make setup and loop live in a different files:

https://wokwi.com/projects/330675632640885330

sutaburosu, 2022-05-03:
I think another possible work-around would be to rename any secondary`.ino` files to `.cpp`. This should work the same in the Arduino IDE and Wokwi.

willie68, 2022-05-04:
No, simply renaming will not work, because your are losing the preprocessor part of the arduino ino file.
And yes you dont need the loop or setup functions. Than your on the C way. Works in Arduino IDE, if you don't need the arduino "macros".
I'm a professional developer, so i browse thru my arduino projects, and in 80% of all my sources, i'll have only one .ino file. 
In all other projects, the solution looking for the setup and loop will work. I have only 1 project, where i dont have any setup or loop, but that is a ATTiny85 project.
So i think looking for setup and loop will work in the wokwi env. for 99% of your users.