wokwi/wokwi-features #354: Add Teensy 3.2 to the microcontroller list

NuttyMonk, 2022-05-07:
Not much to say except i'd like the Teensy 3.2 microcontroller to be added to the list of microcontrollers on Wokwi.

Thanks

**Is your feature request related to a problem? Please describe.**
I can't simulate with a Teensy 3.2 microcontroller.

**Describe the solution you'd like**
I'd like to be able to use a Teensy 3.2 in Wokwi.

urish, 2022-05-07:
Thanks! I opened this request [for voting](https://wokwi.com/features?sort=newest#feature-1228487099).

For implementation reference: the Teensy 3.2 uses the MK20DX256 MCU. This is the relevant [reference manual](https://www.pjrc.com/teensy/K20P64M72SF1RM.pdf).

urish, 2022-05-07:
Note: the scope of adding a new microcontroller is huge. It's usually several weeks of work just to get a simple program running (here's an example from [the journey to emulate the RP2040](https://www.youtube.com/watch?v=POCUbWhBXVA&list=PLLomdjsHtJTxT-vdJHwa3z62dFXZnzYBm&index=1)). 

By default, we only implement the basic peripherals: GPIO, UART, and some of the timer/clock features that are required for functions like `delay()` to work.  If you there are any other peripherals that are show-stopper for you (e.g. I2C), please mention them in this thread, so we'll make an effort to include them as well.

leftCoast, 2022-06-16:
On the teensy you do have a possible resource. Paul S.. something. He's the one that made the thing. And, he has a group of dedicated users that help out. If you got in touch with them, they might think it's a good ice and help?

ChrisStokes, 2022-10-18:
teensy flextimer is super usefu_ as imp_emented by TeensyTimerToo_ _ibrary.

closed 2025-11-03 by urish