wokwi/wokwi-features #357: RGB LED Colors

Quinn135, 2022-05-10:
**Describe the bug**
RGB LEDs do not create the red green and blue, and instead for red, makes a cyan, green makes a yellow, and blue makes a pink/purple.

**To Reproduce**
Simply setup a project, I did it with an arduino uno, and hook up an rgb led. Activate the different parts of it. I have made one for you to look at here : https://wokwi.com/projects/331307612463170131

**Expected behavior**
What should happen is that the R wire should turn on a red color, B should make blue, and G should make green.

**Environment (please complete the following information):**
 - OS: Chrome OS
 - Browser Chrome
 - Version not sure

urish, 2022-05-10:
Seems like a connection issue?

The pins in diagram.json:

```json
    [ "rgb1:R", "uno:13", "green", [ "v0" ] ],
    [ "rgb1:G", "uno:12", "green", [ "v0" ] ],
    [ "rgb1:B", "uno:11", "green", [ "v0" ] ],
```

but in the code G/B seem mixed-up:
```cpp
#define R 13
#define B 12
#define G 11
```

Also, the RGB LED is common anode [by default](https://docs.wokwi.com/parts/wokwi-rgb-led#pin-names), means that the logic is inverted. You have to define a pin as LOW to turn the corresponding LED on.

You can either change the logic in the code, or define the RGB LED as common cathode, as in the following example (which also fixes the mixed-up pins):

https://wokwi.com/projects/331308636705915475

Quinn135, 2022-05-11:
For anyone who didn't know how to change the common to cathode: diagram.json 
[image]

closed 2022-05-11 by Quinn135