wokwi/wokwi-features #359: Mfrc522 Rfid Reader

Khalifla, 2022-05-20:
It would help simulate RFID projects.

urish, 2022-05-20:
Thanks! Opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1242782423)

martinius96, 2023-02-11:
Bro work on it. 9 months later and no progress

urish, 2023-02-11:
Sure, just make sure it gets to the [top of the list](https://wokwi.com/features) first

martinius96, 2023-02-12:
Yep boss. For that 1100 points top "feature" hasn't been implemented for more than year, when it was on first position. 
Let people send you more money and maybe at 10000 points u start doing on it.

urish, 2023-02-12:
Maybe :)

leftCoast, 2023-02-28:
Looks like you were priced out of the market. That's life.

-jim lee

MarioRogue, 2023-04-07:
Is that actually going to happen ?

martinius96, 2023-04-07:
No, because u need to pay 500 USD to have any chance of implementing this feature.

AbdennourBELKACEM, 2024-11-06:
come on guy ,we neeed the rfid module ,  after all this time with no progress ,i think the dev guys are sleeping

martinius96, 2024-11-06:
From time to time I browse Google for RFID Wokwi projects.
There are some, also compilation is running, custom module for RFID RC522 is there, but I don't know how to use it.
There is no guide on "How to attach a tag". Maybe it works in some way that I haven't found yet.

Examples I found:
https://wokwi.com/projects/389180983211282433

https://wokwi.com/projects/395416055497496577

djedu28, 2024-11-07:
Hi @martinius96, I reviewed the links you presented, none of these projects have functional implementation of the chip.

1. The first link started implementing, but didn't complete the SPI communication on the chip. Maybe someone can help with that. Mfrc522 uses SPI communication. Here's the documentation for customizing chips with SPI: https://docs.wokwi.com/chips-api/spi

2. The second link, the chip code doesn't do anything.

-----

As an alternative to `CHIP RC522`, I made my own [`CHIP RDM6300`](https://github.com/djedu28/wokwi-rdm6300-custom-chip/blob/main/docs/README.md), you can use it

> In the past I had the need to emulate an RFID reader, and I did a basic simulation of the RDM6300 Chip https://github.com/djedu28/wokwi-rdm6300-custom-chip
> [image]
  
> The RDM6300 chip uses serial communication.
> I implemented 3 controls. 
> [image]
>
> * Control-1 > "1-Aproximate" or "0-Away" the tag
> * Control-2 > Card1: "1-On" and "0-Off" tag 1
> * Control-3 > Card2: "1-On" and "0-Off" tag 2
 
> When Card1=0 and Card2=1, the chip approximates or moves away tag 2 (Card2), any other configuration uses tag 1

> Automatically translated.
-----
Original message (pt-BR):

Olá @martinius96, revisei os links que você apresentou, nenhum destes projetos possuem a implementação funcional do chip
>
>1. O primeiro link iniciou a implementação, contudo não foi realizado a comunicação SPI no chip. Talvez alguém possa dar comunidade. A comunicação do Mfrc522 é SPI. Aqui possui a documentação da personalização de chips com SPI <https://docs.wokwi.com/chips-api/spi>
>
>2. O segundo link, o código do chip não faz nada.
>
>-----
>
>Como alternativa ao `CHIP RC522`, fiz o meu próprio [`CHIP RDM6300`](https://github.com/djedu28/wokwi-rdm6300-custom-chip/blob/main/docs/README.md), você pode usá-lo
>
>> No passado tive a necessidade de emular um leitor de RFID, e fiz uma simulação básica do RDM6300 Chip <https://github.com/djedu28/wokwi-rdm6300-custom-chip>
>> [image]
>> 
>> o chip RDM6300  usa comunicação serial.
>> implementei 3 controles.
>> [image]
>>
>> * Controle-1 >  "1-aproximar" ou "0-Afastar" a tag
>> * Controle-2 >  Card1: "1-Ativa" e "0-Desativa" a tag 1
>> * Controle-3 >  Card2: "1-Ativa" e "0-Desativa" a tag 2
>
>> Quando a Card1=0 e a Card2=1, o chip aproxima ou afasta a tag 2 (Card2), qualquer outra configuração usa a tag1
>

urish, 2026-02-02:
There we go, the MRFC522 is now officially supported on Wokwi!

Docs: https://docs.wokwi.com/parts/board-mfrc522
Example project: https://wokwi.com/projects/454873133691473921