wokwi/wokwi-features #361: parse bug?

adamfk, 2022-05-27:
**Describe the bug**
There appears to be a parse related bug. The below tiny program throws an exception:

```cpp
#include <stdarg.h>
#include <stdint.h> 

void setup() {
  Serial.begin(115200);
}

void loop() {

  int16_t a = 1, b = 2, c = 3;  
  // "a: %i, b: %i, c: %i,\n", a, b, c);

  Serial.println("Hello!");
  delay(1000);
}
```

```
Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'length')
    at Ja.build (6333-bda6b1d599aa57f0.js:1:169690)
    at async Ja.start (6333-bda6b1d599aa57f0.js:1:172545)
```

**To Reproduce**
https://wokwi.com/projects/332840405861139028

Just try to run the simulation and observe that it does nothing. Open chrome console to see exception above.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: Windows
 - Chrome
 - Version 102.0.5005.63 (Official Build) (64-bit)

urish, 2022-05-27:
Thanks for reporting! Couldn't reproduce here, but made some changes to improve the error reporting. Can you please try again and report back?

adamfk, 2022-05-27:
Every example I try to run now fails with `Build Failed Unknown error`.

I tried running a featured project (not mine): https://wokwi.com/projects/296135008348799496

Note that when I reload the page, I immediately see a `POST https://wokwi.com/build 500` error
[image]

Same outcome in Firefox and Chrome.

urish, 2022-05-28:
Can you check the Network tab of the developers tools and see if you can spot the failing request there?

urish, 2022-05-28:
Never mind that - One of the build servers in the US ran out of disk space. I'm still investigating how to prevent this in the future. 

Can you please try again now?

adamfk, 2022-05-28:
Works! I can no longer reproduce with Chrome or Firefox. Closing.

Thanks for the awesome project! I'll see if I can get my company to become a club member. We are evaluating using wokwi to help with embedded job interviews.

closed 2022-05-28 by adamfk

urish, 2022-05-28:
Thanks! That's an interesting use case. Are there any features that would make it more appealing for this use case ?

adamfk, 2022-05-28:
It's already quite good. I think my main concern is stability. If we are using wokwi to interview candidates, we really need the website to be functioning or have an offline mode. I'd really like the offline mode so that we always have something that we know works. I'll see if I can get my company to join the club and vote for offline mode.

Thanks again!

urish, 2022-06-06:
Thanks! If you're up to it, I'd love to see how you are using Wokwi for interviews. This is the first time I'm seeing this use case :-)

adamfk, 2022-06-08:
I'll send you a link to the project we are using via twitter.

I'll be deleting all the files in that link tomorrow as we just used it for an interview. That's one thing that is a bit of a challenge using it for interviews. I don't want the assignment hanging around or shareable after we use it. It takes a fair amount of effort to create a decent assignment, and we don't want it to be shared on stackoverflow/reddit/...

I realize that once someone edits my project that they basically fork it and that's OK. Maybe we will move the project requirements out of wokwi to limit the amount of assignment info that could be shared.

One other feature that would be nice is time tracking. Being able to see roughly how long someone worked on the project, or the date and times of when they ran the project. Not a big deal, but would be nice.

urish, 2022-06-12:
Thanks for the link, I found the description there hilarious ;-) 🍪

Good note about the time tracking. 

Wokwi compiles the projects using the arduino-cli, building them with the default options. As far as I tested it, it produces the same .hex file as the Arduino IDE does. We're using the latest arduino-cli version (Version: 0.23.0 Commit: 899dc91b Date: 2022-06-06T14:23:44Z), with the latest Arduino code (version 1.8.5).

One idea that comes to mind is that the candidate may had a different board selected in the Arduino IDE. For instance, if they selected an Arduino Mega there, they'd have 256kb of Flash memory, while the Arduino Uno board only has 32kb of Flash memory.