wokwi/wokwi-features #362: Only one tab per function please!

leftCoast, 2022-05-31:
**Is your feature request related to a problem? Please describe.**
The library stuff has two tabs to do its functions. That's one tab too many.

**Describe the solution you'd like**
Lets work on combining this functionality into a single libraries tab.

urish, 2022-05-31:
Thanks! Opened for voting

leftCoast, 2022-07-11:
Maybe even a "hide tab" would work? Default some tabs, .json & library.txt to hidden? Just typing out loud here.

urish, 2022-07-20:
Some more thoughts: We're slowly adding more and more of the diagram.json functionality into the diagram editor. 

So perhaps at some point it'll make sense to have "Simple mode", where the diagram.json/libraries.txt files are hidden, and an Advanced (or Power) Mode, that allows you to edit these files.

leftCoast, 2022-07-20:
That sounds like a good roadmap. The thing is, as long as you have the json thing, I fear this will cause the UI to Lag. And UI is far more important than engineers can possibly understand. (I know, I grew up with engineers.)

urish, 2022-07-20:
I agreed there's a trade-off (as everything in engineering). On the other hand, it allows us to test new features more rapidly: it takes far less time to add a new property in diagram.json in comparison to creating UI.

For instance, the ESP32 has some very specific properties that most users wouldn't really care about (and don't even know about) - these properties are only useful for more advanced users: firmwareOffset, flashSize, psramSize, fullBoot, __timingOptimizations, fastTLS. They are not even documented yet :)

8689-SAMMY, 2023-05-07:
Salve !e buona Domenica ;sono un neo appasionato di programmazione , overo rializzare i progetti con arduino .Ho provato a creare unico conde con piu librerie pero non mi funziona chiedo se qualcuno se mi puo aiutare a fare funzinare ,grazie !allego il code: //Here is a sample code to get started with the NodeMCU ESP8266, TFT_eSPI.h, TFT ILI9341 touchscreen, DS1302 RTC, and XPT2046_touchscreen.h libraries. This code implements a basic clock and touch interface.

//```c++
#include <WiFiClient.h>
#include <WiFi.h>
//#include <TFT_eSPI.h>
#include <XPT2046_Calibrated.h>
#include <RtcDS1302.h>
#include <SparkFunSi4703.h>
#include <PT2314.h>
#include <Adafruit_GFX.h>
#include <Wire.h>
#include <SPI.h>

// Pin definitions
#define TFT_CS     D8   // Pin for TFT chip select
#define TFT_DC     D3   // Pin for TFT data/command
#define TFT_RST    D4   // Pin for TFT reset
#define SI4703_RST D1   // Pin for Si4703 reset
#define SI4703_SCL D5   // Pin for Si4703 clock
#define SI4703_SDA D6   // Pin for Si4703 data
#define SI4703_SEN D7   // Pin for Si4703 interrupt

// Touchscreen pins

#define TS_CS_PIN   2
#define TS_IRQ_PIN  7

#define TFT_CS_PIN 10
#define TFT_DC_PIN 17

// The TIRQ interrupt signal must be used for this example.
#define TIRQ_PIN  2

#define CS_PIN  8
// MOSI=11, MISO=12, SCK=13

// Initialize the TFT and Si4703 objects
//si4703_Breakout(const uint8_t&, const uint8_t&);

int channel;
int volume;
char rdsBuffer[10];

// Initialize the TFT display using the TFT_eSPI library.

// Initialize the RTC using the DS1302 library.

// CONNECTIONS:
// DS1302 CLK/SCLK --> 5
// DS1302 DAT/IO --> 4
// DS1302 RST/CE --> 2
// DS1302 VCC --> 3.3v - 5v
// DS1302 GND --> GND

ThreeWire myWire(4,5,2); // IO, SCLK, CE
RtcDS1302<ThreeWire> Rtc(myWire);

#define countof(a) (sizeof(a) / sizeof(a[0]))

const char data[] = "what time is it";

const int buttonPin = 2;
const int maxChannels = 2; // how many channels do you want? (4 is the physical maximum for PT2314).
//uint8_t channel = 1;
bool loud = false; // this will be toggled later, don't touch.

PT2314 audioswitch;

int freq = 1000; //default frequency
[... 471 lines cut ...]
    // Quadrant 1.
    angle = asin(adjY / hypotenuse);
  } else if (x >= centerX && y >= centerY) {
    // Quadrant 2.
    angle = acos(adjX / hypotenuse) + 0.5 * PI;
  } else if (x < centerX && y >= centerY) {
    // Quadrant 3.
    angle = PI - asin(adjY / hypotenuse);
  } else {
    // Quadrant 4.
    angle = PI + acos(adjX / hypotenuse) + 0.5 * PI;
  }
  return angle * 180 / PI;
}

//***Note: There are some variables and values used in this code that are declared outside of the included code block. Please make sure you provide the necessary values for WIFI_SSID, WIFI_PASSWORD, TP_CS, RTC_DAT, RTC_CLK, and RTC_RST.
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}