wokwi/wokwi-features #363: add stm32 microcontroller

theionSami, 2022-06-14:
i suggest adding stm32 support on wokwi to reach more embedded systems developpers. Thanks!

urish, 2022-06-14:
Thanks for the feedback! Are you using the STM32 yourself? Can you describe which board / MCU are you working with and what development tools are you using?

theionSami, 2022-06-15:
hi Uri,
yes we used stm32 NUCLEO-F401RE before. we used cubeMx back in 2019 but now rather with platformio extension since it offers more features like:
- using vscode accross all embedded dev
- support differents boards (arduino, esp32, stm32..) 
- support arduino framework for cross platforms compilation.
https://docs.platformio.org/en/stable/platforms/ststm32.html

urish, 2022-06-15:
Thanks for the details!

I opened this request for [voting](https://wokwi.com/features?sort=newest#feature-1270975357).

Do you happen this have an example project with platform io + STM32 that you'd like to be able to simulate on Wokwi?

prisar, 2022-06-20:
adding STM32F103C8T6 would be good for me. I am using this mcu

urish, 2022-06-21:
> This is my first chip

Good luck in your journey! What makes this specific chip appealing for you as a first chip for learning?

prisar, 2022-06-22:
Thank you for providing such a convenient platform to experiment various projects online.

>What makes this specific chip appealing for you as a first chip for learning?

I wanted to start with an arm microprocessor because I'm thinking of using some in my future projects. And, it has  ARM Cortex-M3. So, I decided to tinker STM32F103

dzid26, 2023-04-18:
https://wokwi.com/projects/334824244586218068

urish, 2023-05-23:
Initial STM32 support is now available! 

Starting with NUCLEO-C031C6:

* [STM32 HAL Example](https://wokwi.com/projects/365551778332549121)
* [STM32 Arduino Core + I2C LCD](https://wokwi.com/projects/365421666018061313)
* [STM32 Arduino Core + SPI Display](https://wokwi.com/projects/365549388158011393)

Supported peripherals:
- Cortex M0+ (including SysTick, NVIC)
- GPIO
- USART
- SPI
- I2C
- ADC
- RCC (partial)

Missing peripherals:
- CRC
- DBG
- DMA
- EXTI
- IWDG
- PWR
- RTC
- SYSCFG
- TIM1/3/14/16/17
- WWDG

Documentation and new project templates coming soon!

urish, 2023-05-27:
Initial docs for the new Nucleo-C031C6 board: https://docs.wokwi.com/parts/board-st-nucleo-c031c6

urish, 2023-06-11:
Nucleo-L031K6 is now available too: https://docs.wokwi.com/parts/board-st-nucleo-l031k6

urish, 2023-08-14:
L432KC is available is early alpha: https://wokwi.com/projects/364373191510788097

mirkomatontispire, 2023-08-15:
Do you offer paid support/sdk in case we'd like to add custom support to specific board?

brightproject, 2023-09-09:
@urish Is it possible to add a blue (stm32f103) and black (stm32f401) pill to the project?

urish, 2023-09-18:
stm32f103 is already partially implemented: https://wokwi.com/projects/334824244586218068

If we see a good business case, we might add the black pill as well.

brightproject, 2023-09-18:
> If we see a good business case, we might add the black pill as well.
As far as I understand, it is not possible to achieve at least approximate speeds for performing certain operations and simulating SRAM memory from a simulator?
I thought that changing ESP32 [modules](https://docs.wokwi.com/guides/esp32#esp32-boards) changes the execution time of programs on them, but there is no change.
Therefore, adding a black pill in addition to the blue pill is probably pointless.

urish, 2023-09-18:
> As far as I understand, it is not possible to achieve at least approximate speeds for performing certain operations and simulating SRAM memory from a simulator?

It really depends - some things simulate fast, some things simulate slower. There are also certain optimizations that we can employ for certain use cases. Unfortunately, simulation is not a one-size-fits-all solution: there's always a tradeoff between speed, accuracy and complexity.

brightproject, 2023-09-18:
> It really depends - some things simulate fast, some things simulate slower. There are also certain optimizations that we can employ for certain use cases. Unfortunately, simulation is not a [...]
I [would like](https://github.com/wokwi/wokwi-features/issues/641#issuecomment-1698605374) to simulate some things for esp32 and stm32 at at least 1/4 of the real speed.
But in fact, I get a simulation on ESP32 at a speed 10 times lower than on a real STM32.
For example, the [project](https://wokwi.com/projects/374062283976666113) in the simulator is “executed” on ESP32 at a very low speed.
But on a real stm32 it works great with fps of about 16.

urish, 2023-09-25:
@brightproject I looked into the project, and done some initial experimentation. 

It seems like there are two ways to considerably speed up the simulation:
1. Skip the full SPI protocol emulation in your project - I assume it'll give us x2 speedup
2. Use DMA for SPI transactions - I anticipate another speed-up, but it's hard to know how much without trying. 

The first item is a kind of optimization that we have to make available in the simulation engine. Instead of transmitting one bit at a time, we will just transmit whole bytes between the simulated MCU and the ILI9341 module.

The second one is something that has to be done in the code that runs in the simulator. Your code currently uses the CPU to write SPI data directly, byte by byte. Using DMA enables to write larger buffers, so it is likely to generate additional speed up.

Tyrn, 2023-12-05:
But *how* can I start a project based on the Blue Pill? The dropdown menus don't offer it.

urish, 2023-12-05:
> But _how_ can I start a project based on the Blue Pill? The dropdown menus don't offer it.

Blue Pill is not officially supported at the moment. You can try to copy from an existing project, but YMMV.

krr-hws, 2024-10-01:
Many thanks to the developer of the STM32-devices :) . Is it planned to implement support for hardware timer e.g. TIM1 and TIM3 ?

thernstig, 2024-11-05:
@urish any way these could get support? 😃 

- [STM32C071RBT6](https://www.st.com/en/microcontrollers-microprocessors/stm32c071rb.html#sample-buy)
- [NUCLEO-C071RB](https://www.st.com/en/evaluation-tools/nucleo-c071rb.html)

urish, 2024-11-19:
@thernstig sorry, but we have other priorities at the moment.

closed 2026-01-17 by urish