wokwi/wokwi-features #366: Impossible to switch from Arduino to ESP32 mid project? Why?

boga94, 2022-06-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

boga94, 2022-06-15:
I have a phisical project running on ESP32 + LED Matrix, which i was trying to recreate.
I found the led matrix component is missing.
I found a project with the led matrix in it, connected to arduino.
Cannot swap the arduino for ESP32.... BUMMER...
Cannot copy the matrix from one project to the other....     OK.
Cannot create a neopixel strip and copy/paste the whole strip to make a matrix.................
I want a refund.

urish, 2022-06-16:
Thanks for the feedback!

The current way to swap microcontrollers in an existing project is by editing the [diagram.json file](https://docs.wokwi.com/diagram-format). Search for "wokwi-arduino-uno" and replace with "wokwi-esp32-devkit-v1" (you can find the full list of supported boards in the [documentation](https://docs.wokwi.com/diagram-format#parts)).

closed 2022-06-16 by urish