wokwi/wokwi-features #369: Take me to the error!

leftCoast, 2022-06-16:
When discovering an error (Gasp!) in your code. The error window blocks your view youcan't scroll the code and you are left to your own devices to locate said error. 

There is the option of clicking F8 to take you to the fit error.. But who would think of trying that if not told? How about when we click the close button in the error window it just automatically takes us to out first error?

urish, 2022-07-07:
Thanks for the suggestion!

How about now?

urish, 2022-07-07:
Can you post a screen shot?

leftCoast, 2022-07-07:
Of not being able to type?  ok..
/Users/owner/Desktop/cantType.jpg

leftCoast, 2022-07-07:
That didn't seem to work..

urish, 2022-07-07:
It should actually take you there before you click close (as soon as the error pops up):

https://user-images.githubusercontent.com/892318/177871402-09a31806-c474-453a-8cf0-a7a251757226.mp4

Do you get a different experience? Or does it work for you the same way, but it is not what you wanted?

leftCoast, 2022-07-07:
Yes because I often use multiple files. It wont switch the file to locate the error. It does darken **_some_** of the tabs that have errors in them.

urish, 2022-07-20:
>  It does darken **some** of the tabs that have errors in them.

Can you share a link to a project where there some tabs with errors that are not marked?

> And the underlining is a bit.. Unreliable?

How can I reproduce this?

leftCoast, 2022-07-20:
**Can you share a link to a project where there some tabs with errors that are not marked?** I don't think I can. All the tests I did tonight worked fine.

**And the underlining is a bit.. Unreliable?** Doesn't seem to do this anymore. Maybe what I saw was a fluke?

What I'm looking for is for it to, on close, take me to the tab that has the error then to the error itself.

urish, 2022-07-20:
> Wait no! Typing it in exactly like I had it. It has the odd underlining : https://wokwi.com/projects/337690058595238483

Thanks for sharing the link!

Looking at the error message, it looks like the compiler says there's an error there, so Wokwi underlines it:

[image]

leftCoast, 2022-07-20:
OMG! You are 100% correct! Even the IDE fails on this one!

[image]

urish, 2022-07-28:
Update: if the error is in a different file, the editor now automatically jumps to that tab.

closed 2022-09-03 by urish