wokwi/wokwi-features #370: Allow changing compiler flags

HACKER-3000, 2022-06-16:
**Is your feature request related to a problem? Please describe.**
My specific problem right now is that the default implementations of sprintf(), sscanf(), etc. lack support for floats via `%f`. You can change this using compiler flags. Many other useful things can be done using compiler flags aswell.

**Describe the solution you'd like**
A way to configure compiler flags like `{ "type": "wokwi-arduino-uno", "id": "uno", "top": 0, "left": 0, "attrs": {"extra_flags": "-Wl,-u,vfprintf -lprintf_flt -lm"} }`

urish, 2022-06-18:
Thanks for the feedback! Have you tried compiling on your own machine and then loading the hex file into Wokwi?

You can do it by pressing "F1" and choosing "Load HEX File and Start Simulation…"

HACKER-3000, 2022-06-25:
thanks. this should work as an alternative. it isnt as nice as just pressing run but it should work