wokwi/wokwi-features #372: Polish translators from English

Marak123, 2022-06-30:
I translated the json file with English words into Polish equivalents. Rather, everything is fine.
`{
  "userMenu.signIn": "Zaloguj się",
  "userMenu.signUp": "Zarejestruj się",
  "userMenu.discord": "Rozłącz",
  "userMenu.myProjects": "Mój projekt",
  "userMenu.theClub": "Klub",
  "userMenu.featureRoadmap": "Mapa drogowa funkcji",
  "userMenu.logout": "Wyloguj",
  "project.header.save": "Zapisz",
  "project.header.share": "Udostępnij",
  "project.header.likeProjectTooltip": "Polub projekt",
  "project.header.unlikeProjectTooltip": "Cofnij polubienie projektu",
  "project.menu.saveACopy": "Zapisz kopię",
  "project.menu.lockProject": "Zablokuj projekt",
  "project.menu.unlistProject": "Usuń z listy",
  "project.menu.downloadProjectZip": "Pobierz projekt jako zip",
  "project.error.saveFailed": "Błąd zapisu: Projekt jest zablokowany!",
  "project.error.signInToLikeProject": "Zaloguj się, aby polubić projekt.",
  "project.shareDialog.title": "Udostępnij projekt",
  "project.shareDialog.copyToClipboard": "Kopiuj do schowka",
  "project.shareDialog.pleaseSave": "Zapisz projekt, aby udostępnić go.",
  "project.shareDialog.saveProject": "Zapisz projekt",
  "project.shareDialog.closeButton": "Zamknij",
  "simulator.buildFailedDialog.title": "Błąd budowania!",
  "simulator.downloadingDataDialog.title": "Pobieranie danych z karty SD...",
  "editor.fileMenu.rename": "Zmień nazwę",
  "editor.fileMenu.delete": "Usuń",
  "editor.fileMenu.newFile": "Nowy plik...",
  "editor.fileMenu.uploadFile": "Prześlij plik(i)...",
  "editor.fileMenu.confirmDelete": "Plik {{filename}} zostanie usunięty. Czy na pewno chcesz kontynuować?",
  "editor.fileMenu.error.invalidFileName": "Nieprawidłowa nazwa pliku: znak {{char}} nie jest dozwolony.",
  "editor.fileMenu.error.invalidFileType": "Nieprawidłowy typ pliku. Wspierane typy pliku: {{types}}.",
  "editor.fileMenu.error.fileExists": "Plik o takiej nazwie już istnieje.",
  "editor.fileMenu.error.uploadFailed": "Nie udało się przesłać niektórych plików!",
  "editor.sdCard.uploadFiles": "Prześlij pliki",
  "editor.sdCard.uploadCompleteFolder": "Prześlij folder",
  "editor.sdCard.deleteAllFiles": "Usuń wszystkie pliki",
  "editor.sdCard.noFilesYet": "Nie ma plików. Prześlij jakieś pliki.",
  "diagram.menu.fit": "Dopasuj",
  "diagram.menu.zoomIn": "Przybliż",
  "diagram.menu.zoomOut": "Oddal",
  "diagram.menu.toggleGrid": "Przełącz siatkę",
  "diagram.menu.fullScreen": "Pełny ekran",
  "diagram.menu.exitFullScreen": "Zamknij pełny ekran",
  "diagram.menu.help": "Pomoc",
  "diagram.addPart.wokwi-arduino-uno": "Arduino Uno",
  "diagram.addPart.wokwi-arduino-mega": "Arduino Mega 2560",
  "diagram.addPart.wokwi-arduino-nano": "Arduino Nano",
  "diagram.addPart.wokwi-pi-pico": "Raspberry Pi Pico",
  "diagram.addPart.wokwi-attiny85": "ATtiny85",
  "diagram.addPart.wokwi-led": "LED",
  "diagram.addPart.wokwi-pushbutton": "Przycisk",
  "diagram.addPart.wokwi-resistor": "Rerzystor",
  "diagram.addPart.wokwi-slide-switch": "Przełącznik",
  "diagram.addPart.wokwi-membrane-keypad": "Klawiatura",
  "diagram.addPart.wokwi-logic-analyzer": "Logic Analyzer (8 channels)",
  "diagram.addPart.wokwi-max7219-matrix": "LED Dot Matrix (MAX7219)",
  "diagram.addPart.wokwi-led-ring": "NeoPixel Ring",
  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
  "diagram.addPart.wokwi-buzzer": "Brzęczyk",
  "diagram.addPart.wokwi-dht22": "DHT22",
  "diagram.addPart.wokwi-potentiometer": "Potencjometr",
  "diagram.addPart.wokwi-slide-potentiometer": "Potencjometr suwakowy",
  "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
  "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
  "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
  "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
  "diagram.addPart.wokwi-servo": "Serwo",
  "diagram.addPart.wokwi-rotary-dialer": "Dialer obrotowy",
  "diagram.addPart.wokwi-ssd1306": "SSD1306 OLED wyświetlacz",
  "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultradźwiękowy czujnik odległości",
  "diagram.addPart.wokwi-rgb-led": "RGB LED",
  "diagram.addPart.wokwi-7segment": "Wyświetlacz siedmiosegmentowy",
  "diagram.addPart.wokwi-7segment-4digits": "Wyświetlacz siedmiosegmentowy (4 cyfry)",
  "diagram.addPart.wokwi-led-bar-graph": "LED Wykres słupkowy",
  "diagram.addPart.wokwi-analog-joystick": "Joystick analogowy",
  "diagram.addPart.wokwi-ir-receiver": "IR Odbiorca",
  "diagram.addPart.wokwi-ir-remote": "IR Zdalny",
  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Czujnik ruchu",
  "diagram.addPart.wokwi-ntc-temperature-sensor": "Analogowy czujnik temperatury (NTC)",
  "diagram.addPart.wokwi-74hc595": "74HC595 Rejestr zmian",
  "diagram.addPart.wokwi-franzininho": "Franzininho",
  "diagram.addPart.wokwi-photoresistor-sensor": "Czujnik fotorezystora (LDR)",
  "diagram.addPart.wokwi-ky-040": "KY-040 Enkoder obrotowy",
  "diagram.addPart.wokwi-mpu6050": "MPU6050 Akcelerometr + żyroskop",
  "diagram.addPart.wokwi-microsd-card": "Karta microSD"
}`

urish, 2022-07-07:
Thank you! I uploaded the translations, and you can see them live here: https://wokwi.com/projects/new?lang=pl

Mind taking a look and reporting if you can spot any errors?

Marak123, 2022-07-07:
Some elements are not translated even though they have been translated in the .json file, e.g. the display although it has been translated, it still appears as a display or confirmation of the file deletion.

This is a corrected translation (made some small, more precise corrections):

`{
    "userMenu.signIn": "Zaloguj się",
    "userMenu.signUp": "Zarejestruj się",
    "userMenu.discord": "Rozłącz",
    "userMenu.myProjects": "Mój projekt",
    "userMenu.theClub": "Klub",
    "userMenu.featureRoadmap": "Mapa drogowa funkcji",
    "userMenu.logout": "Wyloguj",
    "project.header.save": "Zapisz",
    "project.header.share": "Udostępnij",
    "project.header.likeProjectTooltip": "Polub projekt",
    "project.header.unlikeProjectTooltip": "Cofnij polubienie projektu",
    "project.menu.saveACopy": "Zapisz kopię",
    "project.menu.lockProject": "Zablokuj projekt",
    "project.menu.unlistProject": "Usuń z listy",
    "project.menu.downloadProjectZip": "Pobierz projekt jako zip",
    "project.error.saveFailed": "Błąd zapisu: Projekt jest zablokowany!",
    "project.error.signInToLikeProject": "Zaloguj się, aby polubić projekt.",
    "project.shareDialog.title": "Udostępnij projekt",
    "project.shareDialog.copyToClipboard": "Kopiuj do schowka",
    "project.shareDialog.pleaseSave": "Zapisz projekt, aby udostępnić go.",
    "project.shareDialog.saveProject": "Zapisz projekt",
    "project.shareDialog.closeButton": "Zamknij",
    "simulator.buildFailedDialog.title": "Błąd budowania!",
    "simulator.downloadingDataDialog.title": "Pobieranie danych z karty SD...",
    "editor.fileMenu.rename": "Zmień nazwę",
    "editor.fileMenu.delete": "Usuń",
    "editor.fileMenu.newFile": "Nowy plik...",
    "editor.fileMenu.uploadFile": "Prześlij plik(i)...",
    "editor.fileMenu.confirmDelete": "Plik {{filename}} zostanie usunięty. Czy na pewno chcesz kontynuować?",
    "editor.fileMenu.error.invalidFileName": "Nieprawidłowa nazwa pliku: znak {{char}} nie jest dozwolony.",
    "editor.fileMenu.error.invalidFileType": "Nieprawidłowy typ pliku. Wspierane typy pliku: {{types}}.",
    "editor.fileMenu.error.fileExists": "Plik o takiej nazwie już istnieje.",
    "editor.fileMenu.error.uploadFailed": "Nie udało się przesłać niektórych plików!",
    "editor.sdCard.uploadFiles": "Prześlij pliki",
    "editor.sdCard.uploadCompleteFolder": "Prześlij folder",
    "editor.sdCard.deleteAllFiles": "Usuń wszystkie pliki",
    "editor.sdCard.noFilesYet": "Nie ma plików. Prześlij jakieś pliki.",
    "diagram.menu.fit": "Wyśrodkuj",
    "diagram.menu.zoomIn": "Przybliż",
    "diagram.menu.zoomOut": "Oddal",
    "diagram.menu.toggleGrid": "Przełącz siatkę",
    "diagram.menu.fullScreen": "Pełny ekran",
    "diagram.menu.exitFullScreen": "Zamknij pełny ekran",
    "diagram.menu.help": "Pomoc",
    "diagram.addPart.wokwi-arduino-uno": "Arduino Uno",
    "diagram.addPart.wokwi-arduino-mega": "Arduino Mega 2560",
    "diagram.addPart.wokwi-arduino-nano": "Arduino Nano",
    "diagram.addPart.wokwi-pi-pico": "Raspberry Pi Pico",
    "diagram.addPart.wokwi-attiny85": "ATtiny85",
    "diagram.addPart.wokwi-led": "LED",
    "diagram.addPart.wokwi-pushbutton": "Przycisk",
    "diagram.addPart.wokwi-resistor": "Rerzystor",
    "diagram.addPart.wokwi-slide-switch": "Przełącznik",
    "diagram.addPart.wokwi-membrane-keypad": "Klawiatura",
    "diagram.addPart.wokwi-logic-analyzer": "Analizator logiczny (8 kanałów)",
    "diagram.addPart.wokwi-max7219-matrix": "LED Dot Matrix (MAX7219)",
    "diagram.addPart.wokwi-led-ring": "NeoPixel Pierścień",
    "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
    "diagram.addPart.wokwi-buzzer": "Brzęczyk",
    "diagram.addPart.wokwi-dht22": "DHT22",
    "diagram.addPart.wokwi-potentiometer": "Potencjometr",
    "diagram.addPart.wokwi-slide-potentiometer": "Potencjometr suwakowy",
    "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
    "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
    "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
    "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
    "diagram.addPart.wokwi-servo": "Serwo",
    "diagram.addPart.wokwi-rotary-dialer": "Dialer obrotowy",
    "diagram.addPart.wokwi-ssd1306": "SSD1306 OLED wyświetlacz",
    "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
    "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultradźwiękowy czujnik odległości",
    "diagram.addPart.wokwi-rgb-led": "RGB LED",
    "diagram.addPart.wokwi-7segment": "Wyświetlacz siedmiosegmentowy",
    "diagram.addPart.wokwi-7segment-4digits": "Wyświetlacz siedmiosegmentowy (4 cyfry)",
    "diagram.addPart.wokwi-led-bar-graph": "LED Wykres słupkowy",
    "diagram.addPart.wokwi-analog-joystick": "Joystick analogowy",
    "diagram.addPart.wokwi-ir-receiver": "IR Odbiornik",
    "diagram.addPart.wokwi-ir-remote": "IR Pilot",
    "diagram.addPart.wokwi-pir-motion-sensor": "PIR Czujnik ruchu",
    "diagram.addPart.wokwi-ntc-temperature-sensor": "Analogowy czujnik temperatury (NTC)",
    "diagram.addPart.wokwi-74hc595": "74HC595 Rejestr zmian",
    "diagram.addPart.wokwi-franzininho": "Franzininho",
    "diagram.addPart.wokwi-photoresistor-sensor": "Czujnik fotorezystora (LDR)",
    "diagram.addPart.wokwi-ky-040": "KY-040 Enkoder obrotowy",
    "diagram.addPart.wokwi-mpu6050": "MPU6050 Akcelerometr + żyroskop",
    "diagram.addPart.wokwi-microsd-card": "Karta microSD"
}`

urish, 2022-07-20:
Thanks for the corrections, I uploaded them!

> e.g. the display although it has been translated, it still appears as a display or confirmation of the file deletion.

Good catch, I fixed both! Anything else?

Marak123, 2022-07-20:
If you ask if anything else is "yes".

Untranslated items:
- A4988 Stepper Motor Driver
- DPDT Relay
- Bipolar Stepper Motor
- ILI9341 2.8" TFT-LCD display

To improve:
- "Library Manager" - Translation of the entire tab for managing libraries
- Displaying translations in tooltips (when hovering over an element with the mouse), no translated names are displayed.
- Translation of starting comments in the file "sketch.ino" e.g. comment "// put your setup code here, to run once:"
- Some items that can be added in "Add a new part" are not in the translation

Translation corrected once again:
`{
  "userMenu.signIn": "Zaloguj się",
  "userMenu.signUp": "Zarejestruj się",
  "userMenu.discord": "Rozłącz",
  "userMenu.myProjects": "Mój projekt",
  "userMenu.theClub": "Klub",
  "userMenu.featureRoadmap": "Mapa drogowa funkcji",
  "userMenu.logout": "Wyloguj",
  "project.header.save": "Zapisz",
  "project.header.share": "Udostępnij",
  "project.header.likeProjectTooltip": "Polub projekt",
  "project.header.unlikeProjectTooltip": "Cofnij polubienie projektu",
  "project.menu.saveACopy": "Zapisz kopię",
  "project.menu.lockProject": "Zablokuj projekt",
  "project.menu.unlistProject": "Usuń z listy",
  "project.menu.downloadProjectZip": "Pobierz projekt jako zip",
  "project.error.saveFailed": "Błąd zapisu: Projekt jest zablokowany!",
  "project.error.signInToLikeProject": "Zaloguj się, aby polubić projekt.",
  "project.shareDialog.title": "Udostępnij projekt",
  "project.shareDialog.copyToClipboard": "Kopiuj do schowka",
  "project.shareDialog.pleaseSave": "Zapisz projekt, aby udostępnić go.",
  "project.shareDialog.saveProject": "Zapisz projekt",
  "project.shareDialog.closeButton": "Zamknij",
  "simulator.buildFailedDialog.title": "Błąd budowania!",
  "simulator.downloadingDataDialog.title": "Pobieranie danych z karty SD...",
  "editor.fileMenu.rename": "Zmień nazwę",
  "editor.fileMenu.delete": "Usuń",
  "editor.fileMenu.newFile": "Nowy plik...",
  "editor.fileMenu.uploadFile": "Prześlij plik(i)...",
  "editor.fileMenu.confirmDelete": "Plik {{filename}} zostanie usunięty. Czy na pewno chcesz kontynuować?",
  "editor.fileMenu.error.invalidFileName": "Nieprawidłowa nazwa pliku: znak {{char}} nie jest dozwolony.",
  "editor.fileMenu.error.invalidFileType": "Nieprawidłowy typ pliku. Wspierane typy pliku: {{types}}.",
  "editor.fileMenu.error.fileExists": "Plik o takiej nazwie już istnieje.",
  "editor.fileMenu.error.uploadFailed": "Nie udało się przesłać niektórych plików!",
  "editor.sdCard.uploadFiles": "Prześlij pliki",
  "editor.sdCard.uploadCompleteFolder": "Prześlij folder",
  "editor.sdCard.deleteAllFiles": "Usuń wszystkie pliki",
  "editor.sdCard.noFilesYet": "Nie ma plików. Prześlij jakieś pliki.",
  "diagram.menu.fit": "Wyśrodkuj",
  "diagram.menu.zoomIn": "Przybliż",
  "diagram.menu.zoomOut": "Oddal",
  "diagram.menu.toggleGrid": "Przełącz siatkę",
  "diagram.menu.fullScreen": "Pełny ekran",
  "diagram.menu.exitFullScreen": "Zamknij pełny ekran",
  "diagram.menu.help": "Pomoc",
  "diagram.addPart.wokwi-arduino-uno": "Arduino Uno",
  "diagram.addPart.wokwi-arduino-mega": "Arduino Mega 2560",
  "diagram.addPart.wokwi-arduino-nano": "Arduino Nano",
  "diagram.addPart.wokwi-pi-pico": "Raspberry Pi Pico",
  "diagram.addPart.wokwi-attiny85": "ATtiny85",
  "diagram.addPart.wokwi-led": "LED",
  "diagram.addPart.wokwi-pushbutton": "Przycisk",
  "diagram.addPart.wokwi-resistor": "Rerzystor",
  "diagram.addPart.wokwi-slide-switch": "Przełącznik",
  "diagram.addPart.wokwi-membrane-keypad": "Klawiatura",
  "diagram.addPart.wokwi-logic-analyzer": "Analizator logiczny (8 kanałów)",
  "diagram.addPart.wokwi-max7219-matrix": "LED Dot Matrix (MAX7219)",
  "diagram.addPart.wokwi-led-ring": "NeoPixel Pierścień",
  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
  "diagram.addPart.wokwi-buzzer": "Brzęczyk",
  "diagram.addPart.wokwi-dht22": "DHT22",
  "diagram.addPart.wokwi-potentiometer": "Potencjometr",
  "diagram.addPart.wokwi-slide-potentiometer": "Potencjometr suwakowy",
  "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
  "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
  "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
  "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
  "diagram.addPart.wokwi-servo": "Serwo",
  "diagram.addPart.wokwi-rotary-dialer": "Dialer obrotowy",
  "diagram.addPart.wokwi-ssd1306": "SSD1306 OLED wyświetlacz",
  "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultradźwiękowy czujnik odległości",
  "diagram.addPart.wokwi-rgb-led": "RGB LED",
  "diagram.addPart.wokwi-7segment": "Wyświetlacz siedmiosegmentowy",
  "diagram.addPart.wokwi-7segment-4digits": "Wyświetlacz siedmiosegmentowy (4 cyfry)",
  "diagram.addPart.wokwi-led-bar-graph": "LED Wykres słupkowy",
  "diagram.addPart.wokwi-analog-joystick": "Joystick analogowy",
  "diagram.addPart.wokwi-ir-receiver": "IR Odbiornik",
  "diagram.addPart.wokwi-ir-remote": "IR Pilot",
  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Czujnik ruchu",
  "diagram.addPart.wokwi-ntc-temperature-sensor": "Analogowy czujnik temperatury (NTC)",
  "diagram.addPart.wokwi-74hc595": "74HC595 Rejestr przesówny",
  "diagram.addPart.wokwi-franzininho": "Franzininho",
  "diagram.addPart.wokwi-photoresistor-sensor": "Czujnik fotorezystora (LDR)",
  "diagram.addPart.wokwi-ky-040": "KY-040 Enkoder obrotowy",
  "diagram.addPart.wokwi-mpu6050": "MPU6050 Akcelerometr + żyroskop",
  "diagram.addPart.wokwi-microsd-card": "Karta microSD"
}`

urish, 2022-07-21:
Thanks Marak, I updated the translations file.

It also includes the untranslated items:
* [English JSON](https://wokwi.com/api/i18n/en.json)
* [Up to date Polish JSON with untranslated items](https://wokwi.com/api/i18n/pl.json) - the new items appear as empty strings, e.g.:
   ```json
    "diagram.addPart.wokwi-ili9341": "",
   ```

> "Library Manager" - Translation of the entire tab for managing libraries

Will be done in the future

> Displaying translations in tooltips (when hovering over an element with the mouse), no translated names are displayed.

Where for example?

> Translation of starting comments in the file "sketch.ino" e.g. comment "// put your setup code here, to run once:"

We want to keep the comments in the project in English for now, so when people are sharing code, it'll be readable for people who speak different languages

> Some items that can be added in "Add a new part" are not in the translation

Done!

Marak123, 2022-07-21:
Post fields completed in translation:

`{
  "userMenu.signIn": "Zaloguj się",
  "userMenu.signUp": "Zarejestruj się",
  "userMenu.discord": "Rozłącz",
  "userMenu.myProjects": "Mój projekt",
  "userMenu.theClub": "Klub",
  "userMenu.featureRoadmap": "Mapa drogowa funkcji",
  "userMenu.logout": "Wyloguj",
  "project.header.save": "Zapisz",
  "project.header.share": "Udostępnij",
  "project.header.likeProjectTooltip": "Polub projekt",
  "project.header.unlikeProjectTooltip": "Cofnij polubienie projektu",
  "project.menu.saveACopy": "Zapisz kopię",
  "project.menu.lockProject": "Zablokuj projekt",
  "project.menu.unlistProject": "Usuń z listy",
  "project.menu.downloadProjectZip": "Pobierz projekt jako zip",
  "project.error.saveFailed": "Błąd zapisu: Projekt jest zablokowany!",
  "project.error.signInToLikeProject": "Zaloguj się, aby polubić projekt.",
  "project.shareDialog.title": "Udostępnij projekt",
  "project.shareDialog.copyToClipboard": "Kopiuj do schowka",
  "project.shareDialog.pleaseSave": "Zapisz projekt, aby udostępnić go.",
  "project.shareDialog.saveProject": "Zapisz projekt",
  "project.shareDialog.closeButton": "Zamknij",
  "simulator.buildFailedDialog.title": "Błąd budowania!",
  "simulator.downloadingDataDialog.title": "Pobieranie danych z karty SD...",
  "editor.fileMenu.rename": "Zmień nazwę",
  "editor.fileMenu.delete": "Usuń",
  "editor.fileMenu.newFile": "Nowy plik...",
  "editor.fileMenu.uploadFile": "Prześlij plik(i)...",
  "editor.fileMenu.confirmDelete": "Plik {{filename}} zostanie usunięty. Czy na pewno chcesz kontynuować?",
  "editor.fileMenu.error.invalidFileName": "Nieprawidłowa nazwa pliku: znak {{char}} nie jest dozwolony.",
  "editor.fileMenu.error.invalidFileType": "Nieprawidłowy typ pliku. Wspierane typy pliku: {{types}}.",
  "editor.fileMenu.error.fileExists": "Plik o takiej nazwie już istnieje.",
  "editor.fileMenu.error.uploadFailed": "Nie udało się przesłać niektórych plików!",
  "editor.sdCard.uploadFiles": "Prześlij pliki",
  "editor.sdCard.uploadCompleteFolder": "Prześlij folder",
  "editor.sdCard.deleteAllFiles": "Usuń wszystkie pliki",
  "editor.sdCard.noFilesYet": "Nie ma plików. Prześlij jakieś pliki.",
  "diagram.menu.fit": "Wyśrodkuj",
  "diagram.menu.zoomIn": "Przybliż",
  "diagram.menu.zoomOut": "Oddal",
  "diagram.menu.toggleGrid": "Przełącz siatkę",
  "diagram.menu.fullScreen": "Pełny ekran",
  "diagram.menu.exitFullScreen": "Zamknij pełny ekran",
  "diagram.menu.help": "Pomoc",
  "diagram.addPart.board-esp-01": "ESP-01 WiFi Modem",
  "diagram.addPart.board-ssd1306": "SSD1306 OLED wyświetlacz",
  "diagram.addPart.wokwi-74hc165": "74HC165 Wejściowy Rejestr przesówny (PISO)",
  "diagram.addPart.wokwi-74hc595": "74HC595 Rejestr przesówny",
  "diagram.addPart.wokwi-7segment": "Wyświetlacz siedmiosegmentowy",
  "diagram.addPart.wokwi-7segment-4digits": "Wyświetlacz siedmiosegmentowy (4 cyfry)",
  "diagram.addPart.wokwi-a4988": "A4988 Sterownik silnika krokowego",
  "diagram.addPart.wokwi-analog-joystick": "Joystick analogowy",
  "diagram.addPart.wokwi-buzzer": "Brzęczyk",
  "diagram.addPart.wokwi-dht22": "DHT22",
  "diagram.addPart.wokwi-dip-switch-8": "Przełącznik DIP 8",
  "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultradźwiękowy czujnik odległości",
  "diagram.addPart.wokwi-ili9341": "ILI9341 2.8\" TFT-LCD wyświetlacz",
  "diagram.addPart.wokwi-ir-receiver": "IR Odbiornik",
  "diagram.addPart.wokwi-ir-remote": "IR Pilot",
  "diagram.addPart.wokwi-ks2e-m-dc5": "DPDT Przekaźnik",
  "diagram.addPart.wokwi-ky-040": "KY-040 Enkoder obrotowy",
  "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
  "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
  "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
  "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
  "diagram.addPart.wokwi-led": "LED",
  "diagram.addPart.wokwi-led-bar-graph": "LED Wykres słupkowy",
  "diagram.addPart.wokwi-led-ring": "NeoPixel Pierścień",
  "diagram.addPart.wokwi-logic-analyzer": "Analizator logiczny (8 kanałów)",
  "diagram.addPart.wokwi-max7219-matrix": "LED Dot Matrix (MAX7219)",
  "diagram.addPart.wokwi-membrane-keypad": "Klawiatura",
  "diagram.addPart.wokwi-microsd-card": "Karta microSD",
  "diagram.addPart.wokwi-mpu6050": "MPU6050 Akcelerometr + żyroskop",
  "diagram.addPart.wokwi-neopixel": "NeoPixel",
  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
  "diagram.addPart.wokwi-nlsf595": "NLSF595 SPI Trójkolorowy sterownik LED",
  "diagram.addPart.wokwi-ntc-temperature-sensor": "Analogowy czujnik temperatury (NTC)",
  "diagram.addPart.wokwi-photoresistor-sensor": "Czujnik fotorezystora (LDR)",
  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Czujnik ruchu",
  "diagram.addPart.wokwi-potentiometer": "Potencjometr",
  "diagram.addPart.wokwi-pushbutton": "Przycisk",
  "diagram.addPart.wokwi-resistor": "Rerzystor",
  "diagram.addPart.wokwi-rgb-led": "RGB LED",
  "diagram.addPart.wokwi-rotary-dialer": "Dialer obrotowy",
  "diagram.addPart.wokwi-servo": "Serwo",
  "diagram.addPart.wokwi-slide-potentiometer": "Potencjometr suwakowy",
  "diagram.addPart.wokwi-slide-switch": "Przełącznik",
  "diagram.addPart.wokwi-stepper-motor": "Dwubiegunowy silnik krokowy",
  "diagram.addPart.wokwi-tv": "PAL TV"
}`

> > Displaying translations in tooltips (when hovering over an element with the mouse), no translated names are displayed.

> Where for example?

When you hover the "Start the simulation" button or the "Stop the simulation" button while the simulation is running, the same applies to the "Add new part" button and the button with vertical dots, and the rest of the buttons.

Besides, it would be nice if somewhere on the home page after entering wokwi.com (e.g. in the corner of the page) there would be an option to change the language because I did not see such an option, unless it is skip this suggestion.

urish, 2022-07-24:
> Post fields completed in translation:

Thank you! Updated :)

> Besides, it would be nice if somewhere on the home page after entering wokwi.com (e.g. in the corner of the page) there would be an option to change the language because I did not see such an option, unless it is skip this suggestion.

That's a good point. I'm still considering what's the best spot to insert it. The top bar makes sense, but it's already pretty crowded on mobile:

[image]

We could potentially insert it into the "User" menu, that opens when you click on the Avatar on the right hand side, but that menu doesn't appear unless you are signed in.

Marak123, 2022-07-24:
Maybe it is worth making account settings in which it would be, among others language settings (if it is possible to do account settings), or do this option on the mobile version when you click on the account's avatar (this is where a list pops up where you can choose discord, club, projects, logout, etc.) add an option to change the language there, which will save the selected language in the browser cookies. You can also change the header a little on the mobile version and instead of two buttons (save and share) add one button that will be a list and when you click, these two buttons will appear. And through this maneuver, you can also add another language change button.

urish, 2022-07-28:
Right, I opened a different issue (#391) to track this

urish, 2022-07-28:
> When you hover the "Start the simulation" button or the "Stop the simulation" button while the simulation is running, the same applies to the "Add new part" button and the button with vertical dots, and the rest of the buttons.

Hi Marak, these texts are now available for translation!

Same links as before:

* [English JSON](https://wokwi.com/api/i18n/en.json)
* [Up to date Polish JSON with untranslated items](https://wokwi.com/api/i18n/pl.json) 

All the new ones start with "simulator.main."

Marak123, 2022-07-28:
`{
  "userMenu.signIn": "Zaloguj się",
  "userMenu.signUp": "Zarejestruj się",
  "userMenu.discord": "Rozłącz",
  "userMenu.myProjects": "Mój projekt",
  "userMenu.theClub": "Klub",
  "userMenu.featureRoadmap": "Mapa drogowa funkcji",
  "userMenu.logout": "Wyloguj",
  "project.header.save": "Zapisz",
  "project.header.share": "Udostępnij",
  "project.header.likeProjectTooltip": "Polub projekt",
  "project.header.unlikeProjectTooltip": "Cofnij polubienie projektu",
  "project.menu.saveACopy": "Zapisz kopię",
  "project.menu.lockProject": "Zablokuj projekt",
  "project.menu.unlistProject": "Usuń z listy",
  "project.menu.downloadProjectZip": "Pobierz projekt jako zip",
  "project.error.saveFailed": "Błąd zapisu: Projekt jest zablokowany!",
  "project.error.signInToLikeProject": "Zaloguj się, aby polubić projekt.",
  "project.shareDialog.title": "Udostępnij projekt",
  "project.shareDialog.copyToClipboard": "Kopiuj do schowka",
  "project.shareDialog.pleaseSave": "Zapisz projekt, aby udostępnić go.",
  "project.shareDialog.saveProject": "Zapisz projekt",
  "project.shareDialog.closeButton": "Zamknij",
  "simulator.main.restart": "Uruchom ponownie symulację",
  "simulator.main.start": "Rozpocznij symulację",
  "simulator.main.resume": "Wznów",
  "simulator.main.pause": "Zatrzymaj",
  "simulator.main.addNewPart": "Dodaj nowy element",
  "simulator.main.stopSimulation": "Zatrzymaj symulację",
  "simulator.main.zoomGridTooltip": "Zoom, siatka i więcej",
  "simulator.buildFailedDialog.title": "Błąd budowania!",
  "simulator.downloadingDataDialog.title": "Pobieranie danych z karty SD...",
  "editor.fileMenu.rename": "Zmień nazwę",
  "editor.fileMenu.delete": "Usuń",
  "editor.fileMenu.newFile": "Nowy plik...",
  "editor.fileMenu.uploadFile": "Prześlij plik(i)...",
  "editor.fileMenu.confirmDelete": "Plik {{filename}} zostanie usunięty. Czy na pewno chcesz kontynuować?",
  "editor.fileMenu.error.invalidFileName": "Nieprawidłowa nazwa pliku: znak {{char}} nie jest dozwolony.",
  "editor.fileMenu.error.invalidFileType": "Nieprawidłowy typ pliku. Wspierane typy pliku: {{types}}.",
  "editor.fileMenu.error.fileExists": "Plik o takiej nazwie już istnieje.",
  "editor.fileMenu.error.uploadFailed": "Nie udało się przesłać niektórych plików!",
  "editor.sdCard.uploadFiles": "Prześlij pliki",
  "editor.sdCard.uploadCompleteFolder": "Prześlij folder",
  "editor.sdCard.deleteAllFiles": "Usuń wszystkie pliki",
  "editor.sdCard.noFilesYet": "Nie ma plików. Prześlij jakieś pliki.",
  "diagram.menu.fit": "Wyśrodkuj",
  "diagram.menu.zoomIn": "Przybliż",
  "diagram.menu.zoomOut": "Oddal",
  "diagram.menu.toggleGrid": "Przełącz siatkę",
  "diagram.menu.fullScreen": "Pełny ekran",
  "diagram.menu.exitFullScreen": "Zamknij pełny ekran",
  "diagram.menu.help": "Pomoc",
  "diagram.addPart.board-esp-01": "ESP-01 WiFi Modem",
  "diagram.addPart.board-ssd1306": "SSD1306 OLED wyświetlacz",
  "diagram.addPart.wokwi-74hc165": "74HC165 Wejściowy Rejestr przesówny (PISO)",
  "diagram.addPart.wokwi-74hc595": "74HC595 Rejestr przesówny",
  "diagram.addPart.wokwi-7segment": "Wyświetlacz siedmiosegmentowy",
  "diagram.addPart.wokwi-7segment-4digits": "Wyświetlacz siedmiosegmentowy (4 cyfry)",
  "diagram.addPart.wokwi-a4988": "A4988 Sterownik silnika krokowego",
  "diagram.addPart.wokwi-analog-joystick": "Joystick analogowy",
  "diagram.addPart.wokwi-buzzer": "Brzęczyk",
  "diagram.addPart.wokwi-dht22": "DHT22",
  "diagram.addPart.wokwi-dip-switch-8": "Przełącznik DIP 8",
  "diagram.addPart.wokwi-ds1307": "DS1307 RTC",
  "diagram.addPart.wokwi-hc-sr04": "HC-SR04 Ultradźwiękowy czujnik odległości",
  "diagram.addPart.wokwi-ili9341": "ILI9341 2.8\" TFT-LCD wyświetlacz",
  "diagram.addPart.wokwi-ir-receiver": "IR Odbiornik",
  "diagram.addPart.wokwi-ir-remote": "IR Pilot",
  "diagram.addPart.wokwi-ks2e-m-dc5": "DPDT Przekaźnik",
  "diagram.addPart.wokwi-ky-040": "KY-040 Enkoder obrotowy",
  "diagram.addPart.wokwi-lcd1602": "LCD 16x2",
  "diagram.addPart.wokwi-lcd1602-i2c": "LCD 16x2 (I2C)",
  "diagram.addPart.wokwi-lcd2004": "LCD 20x4",
  "diagram.addPart.wokwi-lcd2004-i2c": "LCD 20x4 (I2C)",
  "diagram.addPart.wokwi-led": "LED",
  "diagram.addPart.wokwi-led-bar-graph": "LED Wykres słupkowy",
  "diagram.addPart.wokwi-led-ring": "NeoPixel Pierścień",
  "diagram.addPart.wokwi-logic-analyzer": "Analizator logiczny (8 kanałów)",
  "diagram.addPart.wokwi-max7219-matrix": "LED Dot Matrix (MAX7219)",
  "diagram.addPart.wokwi-membrane-keypad": "Klawiatura",
  "diagram.addPart.wokwi-microsd-card": "Karta microSD",
  "diagram.addPart.wokwi-mpu6050": "MPU6050 Akcelerometr + żyroskop",
  "diagram.addPart.wokwi-neopixel": "NeoPixel",
  "diagram.addPart.wokwi-neopixel-meter": "NeoPixel Meter",
  "diagram.addPart.wokwi-nlsf595": "NLSF595 SPI Trójkolorowy sterownik LED",
  "diagram.addPart.wokwi-ntc-temperature-sensor": "Analogowy czujnik temperatury (NTC)",
  "diagram.addPart.wokwi-photoresistor-sensor": "Czujnik fotorezystora (LDR)",
  "diagram.addPart.wokwi-pir-motion-sensor": "PIR Czujnik ruchu",
  "diagram.addPart.wokwi-potentiometer": "Potencjometr",
  "diagram.addPart.wokwi-pushbutton": "Przycisk",
  "diagram.addPart.wokwi-resistor": "Rerzystor",
  "diagram.addPart.wokwi-rgb-led": "RGB LED",
  "diagram.addPart.wokwi-rotary-dialer": "Dialer obrotowy",
  "diagram.addPart.wokwi-servo": "Serwo",
  "diagram.addPart.wokwi-slide-potentiometer": "Potencjometr suwakowy",
  "diagram.addPart.wokwi-slide-switch": "Przełącznik",
  "diagram.addPart.wokwi-stepper-motor": "Dwubiegunowy silnik krokowy",
  "diagram.addPart.wokwi-tv": "PAL TV"
}`

Added translation elements.

closed 2022-12-14 by urish