wokwi/wokwi-features #375: Add different sizes of the breadboard element (for example half size, mini) and make the + and - rows on the top and bottom usable

MarcelskyXD, 2022-07-02:
**Is your feature request related to a problem? Please describe.**
Yes, + and - rows dont do anything, infact they dont even have pins on them, also different breadboard sizes would be good.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Make the + and - rows work on breadboards and also add different sizes.

urish, 2022-07-07:
Thanks for the suggestion! We first need to complete wokwi/wokwi-elements#31, so I added a comment there with reference to this.

In general, the [breadboard feature](https://wokwi.com/features#feature-650859885) got some votes already. If it keeps getting votes and climbing up the list, this will happen sooner.

closed 2022-07-07 by urish

urish, 2022-11-23:
Half breadboard: https://wokwi.com/projects/349109488710582866