wokwi/wokwi-features #377: Add Arduino UNO WIFI Rev2

marzvrover, 2022-07-03:
Feature request to add support simulating the [Ardunio UNO WIFI Rev2](https://store.arduino.cc/products/arduino-uno-wifi-rev2)

urish, 2022-07-07:
Thanks for the feature request! I opened it for [voting](https://wokwi.com/features?sort=newest#feature-1292415201)

d3lta-v, 2023-01-09:
Hi! 

I would like to weigh in with some of my personal experiences developing and using a platform that has a near-identical architecture to the Uno WiFi Rev 2. It is an 8-bit ATmega4809 combined with an ESP32 communicating over the SPI bus, with the 8-bit micro acting as the controller and ESP32 as the peripheral. 

I suspect that this architecture will depend on the [multi-MCU feature](https://github.com/wokwi/wokwi-features/issues/186) being completed first, unless there's a way to emulate just the networking functionality. Since the Uno WiFi Rev2 uses the [WiFiNINA](https://github.com/arduino-libraries/WiFiNINA) library which offers a somewhat (programmatically) similar interface to ESP32's Arduino WiFi library, it could be possible to emulate just the network side of things without actually having to emulate the ESP32 itself, if this feature were to be implemented before the multi-MCU feature is completed. 

A quick and dirty comparison of the difference between ESP32's own WiFi implementation and the WiFiNINA library can be seen here: 

* ESP32: https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/WiFiSTA.h
* WiFiNINA: https://github.com/arduino-libraries/WiFiNINA/blob/master/src/WiFi.h

Phetdanai, 2023-12-22:
To download
Click [this](https://github.com/wokwi/wokwi-features/files/13748702/WiFiNINA-master.zip).

closed 2025-11-03 by urish