wokwi/wokwi-features #379: Duty cycle is halved using LEDC PWM on ESP32-C3

DurandA, 2022-07-05:
**Describe the bug**
PWM signals generated by the [LEDC peripheral](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html) have half of the expected duty cycle on the ESP32-C3 (and not on ESP32).

**To Reproduce**
[Simulation project](https://wokwi.com/projects/336380083260359252) with logic analyzer using the [official LEDC example](https://github.com/espressif/esp-idf/blob/release/v5.0/examples/peripherals/ledc/ledc_basic/main/ledc_basic_example_main.c).

**Expected behavior**
GPIO5/D0 should have a duty cycle of 50% instead of 25% in the above example.

**Environment (please complete the following information):**
 - OS: Linux Mint 20.3
 - Browser: Firefox 101

Dlloydev, 2022-07-06:
Looks to me like the bug is in the example itself which predates their latest release [Arduino ESP32 Version 2.0.3](https://github.com/espressif/arduino-esp32/releases) which has fixes on LEDC.

Changing line 27 in the example to `.duty_resolution  = LEDC_DUTY_RES - 1,` fixes the issue and seems to be consistent with their [LEDC ESP32-C3 reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/ledc.html) that states:

> The range of the duty cycle values passed to functions depends on selected duty_resolution and should be from 0 to (2 ** duty_resolution) - 1. For example, if the selected duty resolution is 10, then the duty cycle values can range from 0 to 1023. This provides the resolution of ~0.1%.

I have a library that works with LEDC PWM [here](https://github.com/Dlloydev/ESP32-ESP32S2-AnalogWrite) and has the following examples where the duty is as expected:
Wokwi phase shift examples ... [ESP32](https://wokwi.com/projects/334722465700774482) ... [ESP32-S2](https://wokwi.com/projects/334765722024542804) ... [ESP32-C3](https://wokwi.com/projects/334856585002091092)

Note that in [pwmWrite.cpp](url), I set the bit resolution as follows:

```c++
#if (CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3 || CONFIG_IDF_TARGET_ESP32C3)
      bits = ((bits - 1) & 0xF);
#endif
```

DurandA, 2022-07-06:
> Looks to me like the bug is in the example itself which predates their latest release [Arduino ESP32 Version 2.0.3](https://github.com/espressif/arduino-esp32/releases) which has fixes on LEDC.

If you try the exact same Wokwi project with an ESP32 instead of an ESP32-S3, then the measured duty cycle is 50% as expected.

urish, 2022-07-06:
Thanks for the super detailed bug report!

I can confirm this issue reproduces. This is the output I see in the simulator:

[image]

And this is the output I can see running the same binary (obtained by pressing "F1" -> "Download Compiled HEX Firmware") on a physical device:

[image]

To summarize: In the simulator, the signal goes high for 100 us, low for 300 us. Where as on the physical device, it goes high for 100 us, low for 100 us.

urish, 2022-07-06:
Uploaded a fix - can you please try again now and report?

DurandA, 2022-07-06:
The fix works fine.

Thank you for your awesome project.

closed 2022-07-06 by DurandA