wokwi/wokwi-features #381: Wrong SPI signals on ESP32

DurandA, 2022-07-07:
**Describe the bug**
I wrote a firmware which sends SPI transactions (TX only) using `tx_data` (not `tx_buffer`). However, the `CLK` and `MOSI` signals are missing and the `CS` signal seems to be the `MOSI` signal. On real hardware, these signals are not missing.

**To Reproduce**
[Simulation project](https://wokwi.com/projects/336568068560912978) with logic analyzer.

This produces the following trace during the two SPI transactions:
[image]

**Expected behavior**
This is the trace that is produced on a real ESP32:
[image]

urish, 2022-07-08:
Thanks for detailed bug report! It's very useful to have the project link, together with the real device vs simulator traces.

The issue has to do with a missing part in the I/O Matrix. The SPI signals are currently always routed to the default I/O pins: miso=12, mosi=13, sck=14 for SPI2.

For most users this setup is good enough, but definitely not for your use case.

Out of curiosity, what are you using the simulator for?

DurandA, 2022-07-08:
> The issue has to do with a missing part in the I/O Matrix. The SPI signals are currently always routed to the default I/O pins: miso=12, mosi=13, sck=14 for SPI2.

Thank you for the clarification. Now I see that [IOMUX is partially implemented](https://docs.wokwi.com/guides/esp32#simulation-features) on the documentation. Btw, I see on the same table that RNG is not available ESP-C3 chips which is not the case.

Should I close the issue?

> Out of curiosity, what are you using the simulator for?

I'd like to add a shift register to control a 7 segments display to [an existing hobby project](https://github.com/DurandA/reflow946) in order to free some pins (enabling switching from the ESP32 to the ESP32-C3 in the process).

Thank you again for making this awesome simulator.

urish, 2022-07-10:
> Should I close the issue?

Let's keep it open as the IOMUX is coming in a few weeks, so we can re-test this once the IOMUX is in place.

> Btw, I see on the same table that RNG is not available ESP-C3 chips which is not the case.

Good catch, fixed!

> I'd like to add a shift register to control a 7 segments display to [an existing hobby project](https://github.com/DurandA/reflow946) in order to free some pins

Cool project! I used to have that hot plate in the past.  
I can imagine at least one project where configuring the profiles could have been useful

urish, 2023-03-23:
Update: the IOMUX has been implemented, so this is mostly working now. 

As far as I can tell, MOSI and CLK are now working correctly. CS is still missing as we don't implement hardware chip select for ESP32 SPI yet (I'll update once we support it)

Vollbrecht, 2023-03-25:
Thank you for your great simulator and for your work.
Here are also some of my findings. I am currently using the VSCode plugin to simulate my project locally. I am building with the latest rust std_hals from git. 
I can confirm that MOSI / MISO /SCK are only working when not connected through the GPIO matrix  rather using the default SPI pins ( using SPI2 device) on the esp32c3. Default pins here are ( clk: 6 mosi: 7 miso:2 cs:10)
Furthermore i discovered a wired toggle glitch on MOSI / MISO lines shown in the following picture.
[image]

urish, 2023-03-25:
Thanks! 

Regarding the toggle glitch - can you try connecting the MOSI / MISO lines to GND through a resistor (so a pull-down resistor)? I'm guessing that the lines are held in High-Z state while not being used, and in this case, the simulator sees them as floating input pins and randomly assigns them a value (either high or low). If by adding a resistor the glitches are gone, this will confirm my guess.

Vollbrecht, 2023-03-25:
One other problem i found is that the simulation speeds don't add up. 3 days ago wokwi always simulated the bus with an speed of 1MHz. Today wokwi always simulates with an bus speed of 4MHz. ( I set the bus bus speed to 500khz). If i setup the busspeed to 1MHz  the simulation lines up with the logic analyzer. Here is an example picture:
[image]

Vollbrecht, 2023-03-25:
> Thanks!
>
> Regarding the toggle glitch - can you try connecting the MOSI / MISO lines to GND through a resistor (so a pull-down resistor)? I'm guessing that the lines are held in High-Z state while [...]

Ty for the suggestion. I tested it with 3 resistors but this doesn't change it. 
For tracking here is my minimal wokwi example project: https://wokwi.com/projects/359915773075820545

Vollbrecht, 2023-03-25:
one more interesting note is: the simulator always creates an internal loopback from MOSI to MISO it seems. even if i unconnect the line between MOSI and MISO.

urish, 2023-03-25:
> one more interesting note is: the simulator always creates an internal loopback from MOSI to MISO it seems. even if i unconnect the line between MOSI and MISO.

How can I observe this?

Vollbrecht, 2023-03-25:
hmm i just tested it with non default pins (on wokwi.com). and MOSI and CLK are working with non default pins in my test now. GJ !

urish, 2023-03-25:
Thanks! Also, just uploaded a fix for the glitches on MOSI. Can you please have a look again?

(CS and speed yet to be fixed)

Vollbrecht, 2023-03-25:
> Thanks! Also, just uploaded a fix for the glitches on MOSI. Can you please have a look again?
> 
> (CS and speed yet to be fixed)

i changed spi bus speed and rebuild the project. i added again resistors to double check that it was not an connection error and removed them also again. though the toggling glitch is still there. Do i need to reload the tab to get an fixed version of the simulator to refresh ?

urish, 2023-03-25:
Reload the tab - yes

Vollbrecht, 2023-03-25:
in the picture here you can see the spi read buffer contains data even if i didn't connect the MOSI/MISO loopback style. 
[image]
in the second picture the "real hardware output" not in loopback currently.
[image]
if i connect the real hardware in loopback it is the same as in the simulation

Vollbrecht, 2023-03-25:
> Thanks! Also, just uploaded a fix for the glitches on MOSI. Can you please have a look again?
> 
> (CS and speed yet to be fixed)

:tada:  toggle glitch is gone nice one 
[image]

urish, 2023-03-26:
Uploaded a bunch of fixes to the CS issue, MISO issue, and clock speed. Closing as we addressed the original issue - but please test, and if there are still problems, let's open a new issue to address them.

closed 2023-03-26 by urish