wokwi/wokwi-features #382: Long level transitions between GPIO.out_w1ts/out_w1tc writes on ESP32

DurandA, 2022-07-08:
**Describe the bug**
I tried some esp-idf code I wrote for segment displays using a timer and interrupts to switch between digits. This works fine on real ESP32 hardware. However, the display is glitched using Wokwi. We can clearly see a "ghost" of the previously shown digit. It seems to be caused by excessive transition time between `GPIO.out_w1ts` and `GPIO.out_w1tc` writes.

**To Reproduce**
[Wokwi project](https://wokwi.com/projects/336571419230995026) with 7 segments display. `alarm_value` of `timer_set_alarm_value()` is set to a very high number so the transition is visible. Typically, this number should be low in order to see all digits at the same time (using vision persistence).

**Expected behavior**
The interrupt writes GPIO.out_w1ts, GPIO.out_w1tc, GPIO.out_w1ts and GPIO.out_w1tc in quick succession so the transition should be extremely quick and not visible.

urish, 2022-07-08:
Thanks for reporting! 

I don't think it's the transition time, rather the 7-segment display's built-in POV simulation. 

Can you please try adjusting the "fps" attribute of the 7-segment display and see which number seems to give good result? 
The default value is "fps": "80".

DurandA, 2022-07-08:
> I don't think it's the transition time, rather the 7-segment display's built-in POV simulation.

You are absolutely right.

> Can you please try adjusting the "fps" attribute of the 7-segment display and see which number seems to give good result? The default value is "fps": "80".

This improves the display. Not perfect but largely good enough for development. :-)

I currently have a broken wrist so I am very happy to be able to play with your simulator rather than manipulating electronics. Thank you again for your project.

closed 2022-07-08 by DurandA

urish, 2022-07-08:
> This improves the display. Not perfect but largely good enough for development. :-)

Curious to know - what value did you end up using?

> I currently have a broken wrist so I am very happy to be able to play with your simulator rather than manipulating electronics. Thank you again for your project.

Sorry to hear about the broken wrist. I hope it heals quickly!

DurandA, 2022-07-09:
> Curious to know - what value did you end up using?

`"fps": 5000` and an `alarm_value` of 75000. Very flickery, but goof enough to check if it works.

urish, 2022-07-10:
And do you also use alarm_value 75000 with the physical hardware?

DurandA, 2022-07-12:
> And do you also use alarm_value 75000 with the physical hardware?

I use a much smaller alarm_value of 5000 with a FreeRTOS tick rate of 100 Hz on real hardware.

urish, 2022-07-13:
Thanks! So what's the reason for not using the same values with the simulator too?

DurandA, 2022-07-14:
> So what's the reason for not using the same values with the simulator too?

I have to use an incredibly high _fps_ value (e.g. 50000) on the segment display for the ghosting to go away.

I previously noticed that the simulation slowed down significantly with a very high _fps_ value but this no longer seems to be the case. Did you update anything in this regard?

DurandA, 2022-07-14:
Btw, this could easily be fixed on the firmware by writing `GPIO.out_w1ts` and `GPIO.out_w1tc` only once. But I don't think that's the point.

I'm curious to know how does the signal sampling work to drive the 7-segment display. For example—at 60 fps—there is normally very little chance that the signal is sampled between the subsequent `GPIO.out_w1ts` and `GPIO.out_w1tc` writes happening during the same interrupt—especially since the interrupt code has very few instructions.

urish, 2022-07-24:
> Did you update anything in this regard?

Nope

> I'm curious to know how does the signal sampling work to drive the 7-segment display

as a matter of fact, it doesn't do sampling. Rather, duty cycle calculation. It looks for every single change of the pin, and notes the amount of time the pin was low vs high. Then we're taking a running average (that's where FPS come into play), since physical monitors can only display 60 FPS or so, and we also have to take the simulation speed into account.

You can see an early version of the algorithm presented here: https://www.youtube.com/watch?v=7KYzPJ-tQno

Even though, I'm not entirely satisfied with the current algorithm. That's why I'm asking about the values which you use to get the behavior that you are looking for.