wokwi/wokwi-features #385: Add support for PCF8574 and PCF8575 I2C port expanders

HACKER-3000, 2022-07-20:
**Describe the solution you'd like**
Add simulation support for the PCF8574 port expander. Talking to them via I2C relatively simple and probably wont be too hard to implement.

Datasheets: [PCF8574](https://www.nxp.com/docs/en/data-sheet/PCF8574_PCF8574A.pdf) [PCF8575](https://www.ti.com/lit/gpn/PCF8575)

urish, 2022-07-21:
Thanks for the detailed request, I appreciate including the data sheet links.

I opened the request for [voting](https://wokwi.com/features?sort=newest#feature-1311403347).

In the meanwhile, it's possible to implement these chips using the custom chip API (#301).

@arcostasi even did a [proof of concept for the PCF8574](https://wokwi.com/projects/327416882842829395), but the code is outdated and uses an older version of the Custom Chip API, so it needs some extra work before it's usable again.

laldana73, 2022-12-06:
The proof of concept for the PCF8574 is working now [here](https://wokwi.com/projects/350228699564474964)