wokwi/wokwi-features #387: Submit projects for /rust page

urish, 2022-07-27:
Use this issue to submit projects for https://wokwi.com/rust. You can also request to update/replace an existing project, or to remove a project.

SergioGasquez, 2022-07-27:
Template Projects:
- esp32: https://wokwi.com/projects/338154815612781140
- esp32c3: https://wokwi.com/projects/338322025101656660
- esp32c3-rust: https://wokwi.com/projects/338322111264195156
- esp32s2: https://wokwi.com/projects/338154940543271506

urish, 2022-07-27:
Thanks Sergio! I converted these into proper Wokwi templates, and they are now accessible from both the main page and [/rust](https://wokwi.com/rust):

[image]

SergioGasquez, 2022-09-12:
Hello! Since we updated and added new Rust Wokwi builders, here are some interesting projects: 
- Templates:
  - No std: 
    - [[esp32c3-rust | rust | no-std ] template](https://wokwi.com/projects/341957109819114067)
    - [[esp32c3 | rust | no-std ] template](https://wokwi.com/projects/341956530834244180)
    - [[esp32s2 | rust | no-std ] template](https://wokwi.com/projects/341955277702038100)
    - [[esp32 | rust | no-std ] template](https://wokwi.com/projects/341952531613614675)
  - Std:
    -  [[esp32c3-rust | rust | std ] template](https://wokwi.com/projects/338322111264195156)
    - [[esp32c3 | rust | std ] template](https://wokwi.com/projects/338322025101656660)
    - [[esp32s2 | rust | std ] template](https://wokwi.com/projects/338154940543271506)
    - [[esp32 | rust | std ] template](https://wokwi.com/projects/338154815612781140)
- Blinky examples
  - No std: 
    - [[esp32c3 | rust | no-std ] blinky](https://wokwi.com/projects/341956526159692371)
    - [[esp32s2 | rust | no-std ] blinky](https://wokwi.com/projects/341955297469792851)
    - [[esp32 | rust | no-std ] blinky](https://wokwi.com/projects/341952826535051858)
  - Std: 
    - [[esp32c3 | rust | std ] blinky](https://wokwi.com/projects/332188235906155092)
    - [[esp32s2 | rust | std ] blinky](https://wokwi.com/projects/333363982163247699)
    - [[esp32 | rust | std ] blinky](https://wokwi.com/projects/333363577816613460)
- Other projects:
  -  [[esp32c3 | rust | std ] esp-clock](https://wokwi.com/projects/342312626601067091)
  -  [[esp32c3 | rust | no-std ] crispy click](https://wokwi.com/projects/341706650098336338)
  - [[esp32c3 | rust | no-std ] display example](https://wokwi.com/projects/340062891526849108)
  - [[esp32s2 | rust | no-std ] display example](https://wokwi.com/projects/338804923638481490)
  - [[esp32 | rust | no-std ] display example](https://wokwi.com/projects/340062796817367636)

playfulFence, 2022-10-07:
Hi there! We updated the no-std builders with new `max7219-nostd` crate, so here's another example:

[esp32 with led matrix 8x8](https://wokwi.com/projects/344869867332043347)

playfulFence, 2022-10-10:
Hello! We got new (a bit more complicated) example with `max7219` and analog joystick now:

[esp-joystick-draw](https://wokwi.com/projects/342697409287029332)

urish, 2022-10-10:
Thank you! Both are up now

playfulFence, 2022-11-01:
Hi there!
New example is here!
[esp-keypad](https://wokwi.com/projects/346697192934736467)

playfulFence, 2022-11-12:
Hello!
New example with using of beautiful [Samuel's examples](https://github.com/sambenko/esp32s3-box-examples) and new builder is ready

[esp-gallery](https://wokwi.com/projects/348033994448699987)

SergioGasquez, 2022-11-12:
> Hello!
> New example with using of beautiful [Samuel's examples](https://github.com/sambenko/esp32s3-box-examples) and new builder is ready
> 
> [esp-gallery](https://wokwi.com/projects/348033994448699987)

Nice! Although it took me a lot to build the first time (~1min 30), and I assume that dependencies should be already cached

playfulFence, 2022-12-15:
[mipidsi-esp-gallery](https://wokwi.com/projects/351123733116617303)

Vollbrecht, 2023-05-13:
# Wokwi Rust AVR Projects

## Templates

- [Arduino Uno](https://wokwi.com/projects/363998486006827009)
- [Arduino Nano](https://wokwi.com/projects/363998608213131265)
- [Arduino Mega](https://wokwi.com/projects/363998729253409793)

## Examples
- 'Uno'
    - [Rotary Encoder + Stepper Motor using Pin Change Interrupt](https://wokwi.com/projects/363948225373532161)
    - [External Interrupt + Multi LED Blinky + Button Input](https://wokwi.com/projects/363943843393048577)
    - [Serial println (UART) + 16bit Timer1 + Timer1 Compa Interrupt](https://wokwi.com/projects/363955424017849345)

Droog71, 2023-05-27:
Hello, here are some examples I made for various components.
These are all using rust-std-esp.

[ESP32-C3 & HC-SR04](https://wokwi.com/projects/363804767906500609)
[ESP32-C3 Servo Sweep](https://wokwi.com/projects/363882169900584961)
[ESP32-C3, A4988 & Stepper Motor](https://wokwi.com/projects/364699732405947393)
[ESP32 & SSD1306](https://wokwi.com/projects/364989432429386753)

closed 2025-11-03 by urish