wokwi/wokwi-features #391: Button to switch langauges

urish, 2022-07-28:
Create a user interface element that will allow users to switch the language of Wokwi. We already have several translations, but currently there is no way for users to select a different language.

Marak123, 2022-07-28:
Is there maybe some repository with the source code for the wokwi website??

urish, 2022-07-28:
The source code spans over several repositories, some are open - but the website itself is not open.

Why do you ask?

Marak123, 2022-07-28:
I wanted to test with different language change combinations. No need to do this in devtools or page fetch.
Regarding the change of this language, do you want the code or the idea where could it be?

AriellaE, 2022-08-11:
[image]
We added a button to switch language in the user menu.

closed 2022-08-11 by AriellaE