wokwi/wokwi-features #393: chinese translation from English

irongu, 2022-07-29:
I added some new translations for Chinese, except two places.
First is "Unlist project", I don't understand what this menu option mean, so I couldn't find an accurate translation for it.
Second is the "LCD 16x2", I believe that there is an mistake between the "LCD 16x2" and the "LCD 16x2 (I2C)". 
Below is a screenshot of the original file, I think the "source" word should switch places.
[image]
[zh.txt](https://github.com/wokwi/wokwi-features/files/9220122/zh.txt)

urish, 2022-08-01:
Thank you very much! I updated the texts, can you [please have a look](https://wokwi.com/projects/new?lang=zh) and let me know if they are okay now?

About your questions:

> First is "Unlist project", I don't understand what this menu option mean, so I couldn't find an accurate translation for it.

It means - hide the project from the public web site, so only people who have the link can access it. It's similar to how "unlisted videos" on YouTube works.

> Second is the "LCD 16x2", I believe that there is an mistake between the "LCD 16x2" and the "LCD 16x2 (I2C)".

Yes, this was a mistake indeed. I fixed it now.

irongu, 2022-08-02:
It works very well!

For the missing translate:
> It means - hide the project from the public web site, so only people who have the link can access it. It's similar to how "unlisted videos" on YouTube works.

So I think the translation should be like this:
```
project.menu.unlistProject:
- source: Unlist project
- translation: 私有化此项目
```

If there is any new texts need to be translated, I will continue to follow up. Thank you very much for this excellent project!

urish, 2022-08-02:
updated, thank you! 

Closing this issue for now, but feel free to keep posting here new translations as there will be new texts (or open a new issue - both are fine)

closed 2022-08-02 by urish

irongu, 2023-02-23:
I translated new content and optmized some old translations.

[zh.txt](https://github.com/wokwi/wokwi-features/files/10811424/zh.txt)

urish, 2023-02-26:
Thanks! Translations updated

irongu, 2023-08-07:
Chinese translations of new content

[zh.txt](https://github.com/wokwi/wokwi-features/files/12272580/zh.txt)

Besides, I noticed that not all wokwi-docs pages have Chinese translations, how can I help translate wokwi-docs?

urish, 2023-08-07:
Thanks, updated!

To translate wokwi-docs:
1. Here is a quick guide for setting up a local environment for translating: https://github.com/wokwi/wokwi-docs/blob/main/translating.md. This is how @arcostasi (who is working on the Portuguese translation) works.
2. The README also contains links to quickly setup an online development environment (in the cloud): https://github.com/wokwi/wokwi-docs#online-development

And if you have any specific questions, feel free to open an issue on wokwi-docs