wokwi/wokwi-features #398: Raspberry Pi Pico Dual Core support

urish, 2022-08-02:
Currently, the Raspberry Pi Pico simulator only simulates a single core. While this is sufficient for most applications, we've seen some use cases where users want to simulate both cores. One example is using MicroPython with threads:

- https://wokwi.com/projects/338830939016659538

If you have additional use-case examples, feel free to comment on this issue.

urish, 2022-08-02:
Opened for [voting](https://wokwi.com/features?sort=newest#feature-1325417923)

kc8ybv, 2023-03-12:
I would absolutely love to see dual core / multi-threading added to the Pi Pico simulation.

urish, 2023-03-12:
Unfortunately, it hasn't got [any votes] in the past 7 months :-(

hendersonh, 2023-07-23:
I vote also for dual core support

parla84, 2025-07-25:
I vote for multi core support