wokwi/wokwi-features #399: EEPROM persistence for MCU's

Dronki, 2022-08-03:
It's a bit frustrating to have the EEPROM cleared everytime a simulation start, having it persistant would be wonderful and potentially speed up development.
To mitigate this I would suggest that a tab with a table, listing the contents of the EEPROM in a key-value fashion:
|Address|Type|Name|Value|
|--|--|--|--|
|0|float|version|0.12|

The advantages of this would be that we could perhaps modify and more easily debug the content of the EEPROM, and to build even further upon this, import a CSV with the EEPROM-contents.

Another small visual problem opens up with this approach tho, and that if the code is storing a struct in EEPROM, and that could be displayed as a "sub-table" for the entry:
|Address|Type|Name|Value|
|--|--|--|--|
|0|float|version|0.12|
|8|struct|config|[...](#)|
> |Address|Type|Name|Value|
> |--|--|--|--|
> |40|float|version|0.12|

_Clicking the three dots would open up a new table for the struct-data_

**But how would the table know what the structure and content of the struct is?**  
This could be done in several ways, two possible ways would be either to tag the struct in the code like `#pragma eeprom-struct <name>`, or define the struct (once again) in the project json-file.

Please let me know your thoughts, and let me know if I'm vague or my issue is confusing :)

urish, 2022-08-03:
Thanks for the detailed feature request! Opened it for [voting](https://wokwi.com/features?sort=newest#feature-1326931331).

samtal, 2022-10-23:
This feature has been suggested and voted (34 votes) before, with less details. Please combine with the previous one.
(It is essential in my project of remote controls to keep each control's unique code)
Thx

urish, 2022-10-23:
Thanks for pointing out, done!

closed 2022-10-23 by urish