wokwi/wokwi-features #400: Manage wokwi club settings.

leftCoast, 2022-08-04:
The creation of a wokwi club membership is easy and documented, as it should be. Editing your woki club setting or closing your club membership is not. This needs to be addressed.

We need a "Manage wokwi club" page that has these controls on it. AND have these controls added to the documentation. As far as I can tell, they are not.

urish, 2022-08-04:
Hi Jim! Can you elaborate on which controls do you expect to find in the "Manage Wokwi Club" page?

leftCoast, 2022-08-04:
Cost settings per month/per year 
Close account - To round out the user options.

urish, 2022-08-04:
> Cost settings per month/per year

Actually, these custom options are now gone - users have only a single choice between monthly / yearly payment. 

> Close account - To round out the user options

👍

leftCoast, 2022-08-05:
Amount per payment? Or is that fixed now as well?

urish, 2022-08-05:
Fixed now. Simplifying things :)

urish, 2023-07-29:
Done - https://wokwi.com/dashboard/subscriptions

also updated the [docs](https://docs.wokwi.com/getting-started/wokwi-club)

closed 2023-07-29 by urish