wokwi/wokwi-features #402: error using library oled_i2c.zip

jorgechacblogspot, 2022-08-11:
**Describe the bug**

The oled_i2c library allows you to drive a 0.96" 128x64 oled i2c screen with Arduino, it connects to an arduino uno/nano A4 SDA A5 SCL and works ok, but when simulating it it throws an error in wokwi.

**To Reproduce**

Connections:

VCC OLED to 5V Arduino
OLED GND to Arduino GND
SDA OLED to A4 Arduino
SCL OLED to A5 Arduino

In the arduino IDE you must use "add library .zip" and add the OLED_I2C.zip library

Finally the simulation starts in wokwi but the screen is flashing white:
https://wokwi.com/projects/339578927388295763

**Expected behavior**

It was expected to display the message "PRUEBA OLED 128x64" for 5 seconds and then display an image intermittent and indefinitely, which corresponds to the Arduino logo.

**Environment (please complete the following information):**
 
  - OS: [Windows 11]
  - Browser [Google chrome and Microsoft Edge]
  - Versión [update ultimate]

**Additional context**

It was physically tested and both the library and the code run ok, real photos of the operation are attached. I also attach the library "OLED_I2C.Zip" :

[image]
[image]
[image]

[OLED_I2C.zip](https://github.com/wokwi/wokwi-features/files/9311310/OLED_I2C.zip)

urish, 2022-08-11:
Thanks for reporting. I can reproduce the issue (tested the same code on Wokwi vs physical SSD1306)

urish, 2022-08-11:
As a workaround, here's a patched version of the OLED_I2C library that works well with Wokwi:

https://wokwi.com/projects/339728623391998547

jorgechacblogspot, 2022-08-11:
thank you very much mr urish for your time, I imagine that the details are very technical, but could you tell us what the problem was? Would it be necessary in real life to modify the library and recompile it?

urish, 2022-08-12:
Yes, the I2C driver that is built into the library does not wait for completion after sending a stop condition on the I2C bus:

https://github.com/jlegas/OLED_I2C/blob/353b6cf4398c1bc3ebc94c5e766d8a19c3b99eaf/hardware/avr/HW_AVR.h#L91

I'm not sure what the real chip does in this case (need to sit down with a logic analyzer and investigate), but in the simulator, it means that the I2C hardware will still be busy while the library is trying to send the next byte, and thus the next byte will be skipped, and only partial data will be transmitted to the SSD1306. That's why it showed weird things.

As a workaround, I added the following line after sending the stop condition:

```cpp
		while ((TWCR & _BV(TWINT)) == 0) {};										// Wait for TWI to be ready
```

jorgechacblogspot, 2022-08-12:
In that case, should I modify the <HW_AVR.h> file, save the changes and make the zip again, or do I need to recompile the library code?

urish, 2022-08-12:
Yes - but for use in the simulator, you can just use the modified version that I uploaded, just add the following line into the libraries.txt file:

```cpp
OLED_I2C@wokwi:2dbfe51074a0733347f138979ac7b4ba33587875
```

(You can also download the modified .zip from the Library Manager in the project above)

jorgechacblogspot, 2022-08-12:
excellent thanks.

I only have one doubt, the new file oled_i2c.zip weighs only 51.9 kb and the original zip weighed 514 kb

urish, 2022-08-12:
When you upload a custom library, wokwi removes some files (like documentation, images) and keeps only the code - to make things faster

closed 2022-08-23 by urish