wokwi/wokwi-features #403: File names on libraries text or libraries display

DarwinWasWrong, 2022-08-16:
When adding libraries, it is sometimes a frustrating search the for correct file name for the include statements.
A very simple example of this is
LCD IC2 library requires
 #include <LiquidCrystal_I2C.h>
 
libraries.txt shows LiquidCrystal I2C
[image]

Library Manager Shows
[image]

This is the simplest example
Try DHT
[image]

If the library text OR the installed libraries page has the actual library file name included, this would help a lot.

urish, 2022-08-17:
Good idea! Opened [for voting](https://wokwi.com/features?sort=newest#feature-1339676240)

Twistx77, 2022-11-24:
Hi, wouldn't it ever to just use something like the VSCode Integration feature help with this? You could just use Platformio IO to manage the libraries.
The VSCode integration is already working in alfa. Maybe it makes more sense to vote for it rather than this? https://github.com/wokwi/wokwi-features/issues/414

DarwinWasWrong, 2022-12-01:
No, it is a simple change to the UI to allow beginners etc to find what is required to run a library
I don't use VS code

urish, 2025-11-03:
For popular libraries, Wokwi now automatically suggests the library based on your includes, e.g.:

[image]