wokwi/wokwi-features #404: Support different I2C addres on character LCDs PCF8574T

Gjorgjevikj, 2022-08-16:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2022-08-16:
If I understand correctly, the range of valid addresses is 0x20 to 0x27.

What's the use case for changing the address?

Gjorgjevikj, 2022-08-16:
Connecting multiple displays to the same I2C bus. Probably not a most common scenario, but can be usable in certain projects. It does exist as an option on the hardware modules, anyway.

urish, 2022-08-16:
Apparently, it's already supported - use the "i2c-address" attribute to set the address, e.g.:

```json
    {
      "type": "wokwi-lcd2004",
      "id": "lcd2",
      "top": 276.63,
      "left": -16.62,
      "attrs": { "pins": "i2c", "i2c-address": "0x20" }
    }
```

docs: https://docs.wokwi.com/parts/wokwi-lcd1602#attributes

closed 2022-08-16 by urish

Gjorgjevikj, 2022-10-11:
Probably my bad. I did search for that but did not find it on 

https://docs.wokwi.com/parts/wokwi-lcd1602#i2c-configuration

 

Thanks, anyway.

 

BTW, 

although not so common, LCDs with the same Hitachi driver and same connections with different geometries also exist 

8x1, 16x4, 24x2, 40x2, …

Probably it is not to hard to add some of them… just a thought...

urish, 2022-10-11:
> Probably my bad. I did search for that but did not find it on
> https://docs.wokwi.com/parts/wokwi-lcd1602#i2c-configuration

Added a note there!