wokwi/wokwi-features #405: It seems that hd44780.h library is not working rpoperly with Wokwi

Gjorgjevikj, 2022-08-16:
It seems that hd44780.h library is not working properly with Wokwi for i2C connected LCDs
The given sketch that works on hardware does not work on Wokwi

**To Reproduce**
(https://wokwi.com/projects/340166928469328468)

**Expected behavior**
The display connected in the I2C lines using this library is only blinking the backlight
The 2 display (even 3 display setup) works fine when using this library on hardware with the same sketch

**Environment (please complete the following information):**
 WINDOWS 11
Firefox 103.0.2 (64-bit)

Using different LCD libraries, basically the same sketch works on Wokwi
https://wokwi.com/projects/340169847273947731

urish, 2022-08-16:
As far as I can tell, the issue is with how the hd44780_I2Cexp constructor is called. Can you try changing line 40 to:

```cpp
hd44780_I2Cexp  lcd2(0x27, I2Cexp_BOARD_YWROBOT);
```

or the more verbose

```cpp
hd44780_I2Cexp  lcd2(0x27, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH);
```

and report if that works for you?

Gjorgjevikj, 2022-08-16:
hd44780_I2Cexp  lcd2(0x27, I2Cexp_BOARD_YWROBOT);

solves the problem, thanks.

closed 2022-08-16 by urish