wokwi/wokwi-features #413: ILI9341 emulation does not respect RGB pixel order

ivmarkov, 2022-09-05:
**Describe the bug**
The ILI9341 emulator always assumes BGR order, when - in fact - this is configurable via the 0x36 "Memory Access Control" command
(see page 127, section 8.2.29, bit BGR [here](https://docs.rs-online.com/c1eb/0900766b814a504f.pdf).

**To Reproduce**

Difficult to reproduce without too much setup. The reason is, your demo is based on the AdaFruit Arduino library, which - when looking at the code - always requests BGR order with the 0x36 command (bit BGR set = 0x08) as evidenced [here](https://github.com/adafruit/Adafruit_ILI9341/blob/master/Adafruit_ILI9341.cpp#L223) and perhaps you just assume and hard-code to BGR without inspecting that bit.

Unfortunately there are other codebases that request RGB order and [then assume and hard-code and interpret the order as RGB](https://github.com/yuri91/ili9341-rs/blob/master/src/graphics_core.rs#L20), but since the BGR bit seems not to be respected, I get reordered colors (blue instead of red).

Upon request, I can provide a smaller binary for ESP32 & Rust that demos the problem, but I think the issue is pretty straightforward.

**Expected behavior**
The emulator should examine the BGR bit in command 0x36 and interpret the colors appropriately (either as BGR or RGB, depending on what the user had requested).

**Environment (please complete the following information):**
- All, I assume.

**Additional context**
I've seen other issues as well, but too early to comment on those as the root cause is still unclear. Like the CS pin being hard-wired to GPIO15 or else it does not work (why?). Also the performance is pretty bad. But more on these when/if I'm able to nail down the root cause.

urish, 2022-09-05:
Thanks for the detailed report!

The ILI9341 does support the BRG/RGB bit, we added it two months ago. As far as I remember, it was specifically for the rust use case.

I'm not sure why it doesn't work for you, but if you could provide a reproduction that would be helpful in figuring it out.

ivmarkov, 2022-09-05:
Thanks. I'm using something called "Wokwi Embedded Alpha" if that's of any help (basically, a web service that allows me to submit a binary firmware, completely bypassing the Arduino stuff.) 

Is the above running the latest emulation code?

Regardless, I might find a little time tmr to provide a binary + simple sketch. Thanks!

urish, 2022-09-06:
> Is the above running the latest emulation code?

It does

> Regardless, I might find a little time tmr to provide a binary + simple sketch. Thanks again!

This will be very helpful. Or a gitpod workspace with the source code + Wokwi Embedded Alpha setup

urish, 2023-09-29:
Closing as this hasn't been active for over a year.

closed 2023-09-29 by urish