wokwi/wokwi-features #414: IDE (VSCode) Integration

Twistx77, 2022-09-10:
For many people that wants to use Wokwi for a bit more complex projects, working on the browser is less than ideal. I think that instead of reinventing the wheel making a full fledged editor on the browser, a better solution would be to implement some kind of plugging for VSCode or any other IDE or at least an API that would allow to upload the compiled binary file to Wokwi and run the simulation. With the API anyone (Me for example) could create a simple script that could upload the binary and run it. 

This would also be a solution for this feature request, which would not be needed in most cases[Github integration](https://github.com/wokwi/wokwi-features/issues/75#top)
#75

urish, 2022-09-10:
Thanks for the feature request! I opened it for [voting] (https://wokwi.com/features?sort=newest#feature-1368676003).

urish, 2022-11-20:
Good news - we have a prototype version!

https://user-images.githubusercontent.com/892318/202927900-737a5530-9d26-4fef-87a6-6c06a80b5444.mp4

If you want early access, please ping me on [discord](https://wokwi.com/discord).

Twistx77, 2022-11-21:
Bravo!! 👏

This is awesome! This will avoid having to upload the code, solves the keeping repositories up to date etc.

We can work now as with the real device and choose to run it on Wokwi or directly program the real device id available.

Hopefully the debugger can be used throu VSCode in the future and would it make it perfect. I'm glad that I'm supporting the project.

I will try it as soon as I have time.

Thank you!

leon0399, 2022-11-24:
Wow! Impressive work! I'm wondering if the same implementation can be integrated into PlatformIO for simulation.
https://docs.platformio.org/en/latest/advanced/unit-testing/simulators/index.html

urish, 2022-11-24:
Thank you @leon0399! Are you using Platform IO from Visual Studio Code or from command line?

leon0399, 2022-11-24:
> Thank you @leon0399! Are you using Platform IO from Visual Studio Code or from command line?

Both! 😄 
Using in VSCode for development and CLI for scripts and CI

urish, 2022-11-24:
Got it! You can get early access to the VSCode plugin - just ping me on discord. It should work with Platform IO projects (there are even some examples, e.g. [this one](https://github.com/wokwi/arduino-simon-game))

ricardojlrufino, 2022-11-27:
wow, amazing.

steve-barth, 2022-11-29:
Thanks Uri!

urish, 2023-01-15:
Wokwi for VS Code is now in public beta: https://marketplace.visualstudio.com/items?itemName=wokwi.wokwi-vscode

Docs: https://docs.wokwi.com/vscode/getting-started

If you love it, a 5-star rating in the VS Code marketplace would be helpful (but if you don't think it deserves 5 stars, please leave a comment here so we know why). Thanks :)

Twistx77, 2023-01-15:
Done :)

forntoh, 2023-01-27:
Can't install on vscode version 1.70.0

urish, 2023-01-27:
> Can't install on vscode version 1.70.0

What do you see when you are trying to install? Can you please share a screen shot?

forntoh, 2023-01-27:
[image]

urish, 2023-01-27:
Thanks @forntoh! What stops you for updating to the latest VS Code version (1.74.3)?

forntoh, 2023-01-27:
> Thanks @forntoh! What stops you for updating to the latest VS Code version (1.74.3)?

I don't know why I can't update, I don't even see the command to update nor do I find it in the help menu. I'll try again.

forntoh, 2023-01-27:
I don't have that menu item on mine 😄 why?
[image]

urish, 2023-01-27:
Seems it could be related to microsoft/vscode#157383 - where it stopped auto-updating after version 1.70.0. I think the easiest fix is just to download and install the latest version from [the website](https://code.visualstudio.com/Download)

forntoh, 2023-01-27:
> Seems it could be related to [microsoft/vscode#157383](https://github.com/microsoft/vscode/issues/157383) - where it stopped auto-updating after version 1.70.0. I think the easiest fix is just to download and install the latest version from [the website](https://code.visualstudio.com/Download)

Thanks, I found a solution, it's actually related to [vscode#157624](https://github.com/microsoft/vscode/issues/157624) and [vscode#147408](https://github.com/microsoft/vscode/issues/147408)

alextrical, 2023-02-13:
I have noticed that the simulation in the Visual studio environment is a bit blurred, and makes screens difficult to use.
Is this a known bug, or an issue with my setup on Ubuntu?
[image]
Visual Studio Code on left, web version in Firefox on the right

urish, 2023-02-13:
a "bit blurred" is an understatement. The OLED screen is barely usable! 

I haven't seen this issue before. Does it always happen? Which VS Code/ubuntu version?
Does Chrome or Chromium have a similar issue?

alextrical, 2023-02-13:
It seems as if the anti-aliasing is set at the time the simulation is first loaded, and the zoom just increases the pixel size, without recalculating the 'smoothing'. Though it seems that if zoomed in enough it does readjust the image to re readable once zoomed back out
[image]

Current setup is:
Visual studio Code 1.75.1 installed via Snap
KUbuntu 22.10

Chromium does indeed have a bit of an issue, but its definitely useable
[image]

urish, 2023-02-13:
> It seems as if the anti-aliasing is set at the time the simulation is first loaded, and the zoom just increases the pixel size, without recalculating the 'smoothing'. Though it seems that if zoomed in enough it does readjust the image to re readable once zoomed back out

Yes, I've seen this happening with Chrome. I guess it's some sort of optimization that it does to speed up drawing. But never to the degree that you are showing with VS Code. I'll see if I can reproduce it here.

alextrical, 2023-02-13:
https://user-images.githubusercontent.com/35117191/218571618-07228cb8-16a8-4f31-bd71-e14d2eba2cb0.mp4

I will keep an eye out for a fix, as i think this is an issue with my setup and Visual Studio code. If i find a solution to disable the anti-aliasing I will let you know :)
Its not a major issue (and by the sounds of it only effecting me), so I would say don't waste too much time looking into it.

realJoshByrnes, 2023-03-08:
Referencing an issue I'm having using the VSCode extension: https://github.com/wokwi/rp2040js/issues/123

loopj, 2023-06-07:
@alextrical I'm also seeing this exact blurring issue in VSCode, on a mac

alextrical, 2023-06-07:
@loopj In a way I'm glad that there is another instance of this happening on another OS, it will hopefully allow the bug to be tracked down and resolved. :)

pe5er, 2023-12-05:
> I have noticed that the simulation in the Visual studio environment is a bit blurred, and makes screens difficult to use. Is this a known bug, or an issue with my setup on Ubuntu? [image] Visual Studio Code on left, web version in Firefox on the right

I am also having this issue on Windows 11. Other than that, the VSCode Integration is a beast!

FiveTeethless, 2023-12-12:
same here on Windows 11 and VSCode 1.85.0
Chromium is of version 114.0.5735
Let me know, how I can help finding this issue.

Sometimes it renders the view very sharp after some time and fiddeling, but I couldn't find some definitive conditions for it.

pe5er, 2023-12-12:
I found that instead of fixing it by zooming in really far, I could fix it by clicking on the vs code interface outside of the Wokwi simulator, pressing control+ = to zoom in, then control+ - to zoom out. 

This works regardless of how zoomed in the Wokwi simulator itself is.

Hopefully that provides a clue, but failing that it's an easier workaround

urish, 2023-12-17:
Thanks for sharing your workaround @pe5er! I hope I'll be able to reproduce this in a consistent way and then find a solution.

urish, 2023-12-27:
Closing this issue, as Wokwi for VS Code is now out of beta. Feel free to open new issues for any bugs or feature requests related to Wokwi for VS Code.

closed 2023-12-27 by urish

urish, 2024-03-17:
> I found that instead of fixing it by zooming in really far, I could fix it by clicking on the vs code interface outside of the Wokwi simulator, pressing control+ = to zoom in, then control+ - to zoom out.

Good news: I managed to reproduce this, and uploaded a fix that works for me.

@pe5er /  @FiveTeethless / @alextrical / @loopj can you please check if blurring still happens for you?

pe5er, 2024-03-18:
@urish seems fixed for me - thanks for your help!

alextrical, 2024-03-18:
@urish same here, looks good this end too. Thank you so much

FiveTeethless, 2024-03-20:
> Good news: I managed to reproduce this, and uploaded a fix that works for me.
>
> @pe5er / @FiveTeethless / @alextrical / @loopj can you please check if blurring still happens for you?

thanks, also works on my machine! 👍

ShakaAmilcar876, 2024-05-07:
> I have noticed that the simulation in the Visual studio environment is a bit blurred, and makes screens difficult to use. Is this a known bug, or an issue with my setup on Ubuntu? [image] Visual Studio Code on left, web version in Firefox on the right

How do you run custom chips in VSCode? Where do you go to find the 'wokwi_api.h' header file??

urish, 2024-05-12:
> How do you run custom chips in VSCode? Where do you go to find the 'wokwi_api.h' header file??

You can find an example on https://github.com/wokwi/inverter-chip/tree/main/src

Please use the [Wokwi community chat on Discord](https://wokwi.com/discord) for questions.