wokwi/wokwi-features #415: Serial.availableForWrite() returns 0 on Arduino nano board

alonbl, 2022-09-11:
**Describe the bug**

The `Serial.availableForWrite()` returns 0 on Arduino nano board while in esp32 it works correctly.

**To Reproduce**

Consider the following program:

```c++
void setup() {
  Serial.begin(115200);
  Serial.println(Serial.availableForWrite());

}

void loop() {
  Serial.println(Serial.availableForWrite());
}
```

**Expected behavior**

Any number is sufficient... however, in Arduino Nano it always returns 0.
Async programs that rely on the available buffer to wrote are broken.
In esp32 it works correctly.
In real board we get 63.

**Environment (please complete the following information):**

Reproducible at https://wokwi.com/ without any special environment settings.

**Additional context**

N/A

urish, 2022-09-11:
Thanks for reporting!

Are you sure it's always returning 0?

I tested here, and I can see it does initially return larger values:

[image]

The values are slightly different compared to the physical board, which drops to 0 faster:

[image]

alonbl, 2022-09-11:
Thanks for the prompt reply!

It is very strange!!! I can swear that it was zero constant... but now I cannot reproduce this.

I will reopen if I am able to reproduce.

closed 2022-09-11 by alonbl