wokwi/wokwi-features #418: It would be great if we could create esp-idf projects in wokwi

laldana73, 2022-09-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

urish, 2022-09-21:
Hi! You can already create esp-idf projects, e.g.:

* https://wokwi.com/projects/343148817931043411

Is there anything missing?

laldana73, 2022-09-23:
Dear Urish, Thanks for you answer, I was not able to find the option.
Upload File under the Library Manager dropdown.

Thanks

urish, 2022-09-23:
Hello, it is not a part of the library manager. You can edit the project I gave above as a starting point