wokwi/wokwi-features #419: SSD1306 flipHorizontal is ignored

penfold42, 2022-09-22:
**Describe the bug**
The SSD1306 wont flip horizontally (but vertical flip works fine).

**To Reproduce**
Arduino nano with ssd1306 library and it works fine in real life with a 4 pin chinesium display.
Here's my call to the library:

```
void set_contrast_rotation() {
    ssd1306_flipVertical(run_config.display_orientation & 2);
    ssd1306_flipHorizontal(run_config.display_orientation & 1);
}
```

The ssd1306 library code sends this to the display:
```
void ssd1306_flipHorizontal(uint8_t mode)
{
    ssd1306_sendCommand( SSD1306_SEGREMAP | (mode ? 0x00: 0x01 ) );
}
```
**Expected behavior**
Work like it does in real life

**Environment (please complete the following information):**
 - Firefox 104.0.1 (64-bit) Windows 10

**Additional context**
Maybe related to #295 ?

urish, 2022-09-22:
Thanks for reporting! Can you please share a link to a Wokwi project with the complete code that reproduces the issue?

penfold42, 2022-09-23:
It took longer than it should to make this small excerpt from my main project.
I had forgotten that the ssd1306 implements H flip using Set Segment Re-map (A0h/A1h) and this requires repainting the display:
"This command only affects subsequent data input. Data already stored in GDDRAM will have no changes."

Heres my minimal sample:
https://wokwi.com/projects/343460442395378259

Use the serial interface:
 * C=4   Display Contrast (0-9)
 * D=1   Display ON
 * D=0   Display OFF
 * O=0   Display Orientation (0=normal, 1=flipX, 2=flipY, 3=flipXY)

urish, 2022-09-23:
Thanks for spending your time on creating a minimal sample. It was very helpful.

Fix is up - can you please test again?

penfold42, 2022-09-24:
> Thanks for spending your time on creating a minimal sample. It was very helpful.

No probs, time spent up front is almost always a win (for both of us!)

> Fix is up - can you please test again?

Yep - excellent quick work, thank you.
I've tested both the sample and my main project and both display orientation and on/off both work as expected.
Including the need to re-write the buffer for horizontal flip to take effect :)

urish, 2022-09-24:
Excellent!

closed 2022-09-24 by urish