wokwi/wokwi-features #420: SSD1306_SETCONTRAST,  DISPLAYOFF, DISPLAYON not working

penfold42, 2022-09-22:
**Describe the bug**
It appears that setcontrast and displayoff/on dont work.
Are them implemented? I cant see any restrictions listed at https://docs.wokwi.com/parts/board-ssd1306

**To Reproduce**
code that calls these commands appears to have no effect

**Expected behavior**
code that calls these commands should have an effect :)

**Environment (please complete the following information):**
Firefox 104.0.1 (64-bit) Windows 10

**Additional context**

urish, 2022-09-22:
setcontrast - not sure if there's a good reason to support it, but it makes sense to document it.

displayoff/on - can you please share a link to a Wokwi project that uses these?

urish, 2022-09-22:
Yes please! Thanks :

penfold42, 2022-09-23:
https://wokwi.com/projects/343460442395378259

Use the serial interface:
 * C=4   Display Contrast (0-9)
 * D=1   Display ON
 * D=0   Display OFF
 * O=0   Display Orientation (0=normal, 1=flipX, 2=flipY, 3=flipXY)

penfold42, 2022-09-24:
can confirm display on/off now works as per #419 

setcontrast would be a nice to have, but there are plenty of votable ERs here already to keep you busy.
If this were open source, I'd probably just do it as a simple exercise to understand the codebase :)

urish, 2022-09-24:
> If this were open source, I'd probably just do it as a simple exercise to understand the codebase :)

Actually, the code already supports this - it's just not reflected in the UI. But the majority of the work here isn't about code. Here are some other things we need to take into account:

1. Defining what the result will look like (depending on the contrast level).
2. Possibly adding a way for users to either opt-in or opt-out 
3. Testing a bunch of existing projects to see what the display looks like with the new contrast option
4. Documenting the new feature

One of the challenges here is to make sure we're not confusing users - otherwise, we will have to spend time supporting users wondering why their SSD1306 output looks washed out on Wokwi.

I'm not sure if spending a few hours here worth the nice-to-have feature...

penfold42, 2022-09-24:
>     1. Defining what the result will look like (depending on the contrast level).
meh! just scale the current brightness to match the SSD1306 default of 127.
 
>     2. Possibly adding a way for users to either opt-in or opt-out
If you dont use the command, the SSD1306 defaults to 127 on reset.

>     3. Testing a bunch of existing projects to see what the display looks like with the new contrast option
if they look wrong on real hardware, they should look wrong here! :)

>     4. Documenting the new feature
Covered in the datasheet for the SSD1306 as its a standard command

> One of the challenges here is to make sure we're not confusing users - otherwise, we will have to spend time supporting users wondering why their SSD1306 output looks washed out on Wokwi.

Would you rather more complaints from me that it doesnt match the real hardware! :)
 
> I'm not sure if spending a few hours here worth the nice-to-have feature...

agreed - but we should document which SSD1306 commands the emulation supports.