wokwi/wokwi-features #422: SSD1306 vertial flip painting errors - out by 1?

penfold42, 2022-09-24:
**Describe the bug**
I think there's an out by 1 error on either screen painting or rendering.

It looks like 
- row 0 is still at the top of the screen
- row 1 is on the bottom row (when it should be up one)

**To Reproduce**
[](https://wokwi.com/projects/343460442395378259)
Then O=2 and O=0 to flip between Vflip and normal
Normal:
[image]

Vflip:
[image]

**Expected behavior**
Vertical flip should look something like:
[image]

urish, 2022-09-24:
Reproduction: https://wokwi.com/projects/343460442395378259 (same instructions as [here](https://github.com/wokwi/wokwi-features/issues/420#issuecomment-1255732881))

penfold42, 2022-09-24:
> Reproduction: https://wokwi.com/projects/343460442395378259 (same instructions as [here](https://github.com/wokwi/wokwi-features/issues/420#issuecomment-1255732881))

yes, sorry - must have lost it in an edit as i added the pictures.

urish, 2022-09-24:
Fixed! Thanks again for crafting such helpful bug report. It really speeds up the development!

closed 2022-09-24 by urish

penfold42, 2022-09-24:
confirmed, thanks!